nanopb/nanopb #804: pb_decode does not restore tag field when using pb_callback_t in oneof

grimdaly, 2022-08-08:
test.proto:
```
syntax="proto3";

message TestMessage {
    oneof type {
        string text = 1;
        uint32 number = 2;
    }
}
```

test.options:
```
* anonymous_oneof:true
```

I wrote myself a little Test-function that sets up a message. Then the message is encoded and decoded back:

```cpp
#define SEND_TEXT
//#undef SEND_TEXT
void test()
{
    static char text[] = "someTextToTransport";

    /// prepare msg /////////////////////////////////////////////////////////////
    TestMessage messageToEncode = TestMessage_init_zero;
#ifdef SEND_TEXT
    messageToEncode.which_type = TestMessage_text_tag;
    messageToEncode.text.arg = reinterpret_cast<void *>(text);
    messageToEncode.text.funcs.encode = [](pb_ostream_t *stream, const pb_field_iter_t *field,
                                           void *const *arg) {
        const char *text = reinterpret_cast<const char *>(*arg);
        if (!pb_encode_tag_for_field(stream, field)) {
            return false;
        }
        return pb_encode_string(stream, reinterpret_cast<const pb_byte_t *>(text),
                                std::strlen(text));
    };
#else
    static constexpr auto someNumber = 1337;
    messageToEncode.which_type = TestMessage_number_tag;
    messageToEncode.number = someNumber;
#endif

    /// encode /////////////////////////////////////////////////////////////
    std::array<char, 1024> serializedMessage;
    auto outStream = pb_ostream_from_buffer(reinterpret_cast<pb_byte_t *>(serializedMessage.data()),
                                            serializedMessage.size());
    ASSERT_TRUE(pb_encode(&outStream, TestMessage_fields, &messageToEncode));
    const auto messageLength = outStream.bytes_written;

    /// decode /////////////////////////////////////////////////////////////
    std::array<char, 30> receivedText;
    auto inStream = pb_istream_from_buffer(
        reinterpret_cast<const pb_byte_t *>(serializedMessage.data()), messageLength);

    TestMessage messageToDecode = TestMessage_init_zero;
    messageToDecode.text.arg = reinterpret_cast<void *>(receivedText.data());
    messageToDecode.text.funcs.decode = [](pb_istream_t *stream, const pb_field_iter_t *field,
                                           void **arg) -> bool {
        char *receivedText = reinterpret_cast<char *>(*arg);
        if (stream->bytes_left > 30) {
            return false;
        }

        if (!pb_read(stream,
                     const_cast<pb_byte_t *>(reinterpret_cast<const pb_byte_t *>(receivedText)),
                     stream->bytes_left)) {
            return false;
        }
        return true;
    };

    /// decode /////////////////////////////////////////////////////////////
    ASSERT_TRUE(pb_decode(&inStream, TestMessage_fields, &messageToDecode));
#ifdef SEND_TEXT
    ASSERT_EQ(messageToDecode.which_type, TestMessage_text_tag); // ! should be 1 but is 0
#else
    ASSERT_EQ(messageToDecode.which_type, TestMessage_number_tag);
    ASSERT_EQ(messageToDecode.number, someNumber);
#endif
}
```

However the `messageToDecode.which_type` is only correct when i am serializing an integer. Its always 0 when i use string. Am i doing something wrong?
Thank you in advance

PetteriAimonen, 2022-08-08:
Yet another case of "callback inside oneof". An unfortunate interaction of these two features. The oneofs are stored as C union, but it also means that there is no unique space to store the callbacks for fields.

Various workarounds have been made over the years but currently there is no perfect solution.

grimdaly, 2022-08-30:
Okay, thank you. 
The solution for me was to wrap the string in a sub-message and use the submsg_callback feature.

closed 2022-08-30 by grimdaly