nanopb/nanopb #805: [Bazel] cc_nanopb_proto_library() does not find .options file

randyzhg, 2022-08-19:
`cc_nanopb_proto_library` does not find the `.options` file for my proto (it lives in the same directory and has the same name as the `.proto` file). I've found that I can point it to the options file using an absolute path via `extra_protoc_args`, but it seems to only take an absolute path. If `cc_nanopb_proto_library` cannot find options files automatically, is there at least a way to pass options files via relative paths?

What I'm doing now:
```
cc_nanopb_proto_library(
    name = "myproto_nanopb",
    protos = [":myproto_proto"],
    extra_protoc_args = [
        "--nanopb_plugin_opt=-f/asolute/path/to/myproto.options",
    ],
)
```

PetteriAimonen, 2022-08-21:
@silvergasp Do you want to take a look?

silvergasp, 2022-08-24:
Yeah, I'm aware of this one as its something I would also like. I've been experimenting with this a bit. But haven't got anything working yet.

silvergasp, 2022-08-24:
CC @tpudlik

PetteriAimonen, 2022-08-24:
In the SCons rules I solved similar by having the rule take both the .proto and .options files as source files. But I guess because the proto_library rule is shared with other protobuf libraries, that doesn't work so nicely here.

silvergasp, 2022-08-24:
Yeah I think that it should be doable with a new version of those shared protobuf libraries. There is a new attribute [extra_protoc_files](https://rules-proto-grpc.com/en/latest/lang/c.html) that is included in the newer version. So I could see it working something like this.

```python
cc_nanopb_proto_library(
    name = "myproto_nanopb",
    protos = [":myproto_proto"],
    extra_protoc_args = [
        "--nanopb_plugin_opt=-f$(location myproto.options)",
    ],
    extra_protoc_files = ["myproto.options"],
)
```
I haven't tested this approach yet. So it might take a little bit of experimenting.

randyzhg, 2022-08-24:
I just tried using `extra_protoc_files` and it didn't seem to work. `extra_protoc_files` expects a Bazel target because passing the absolute path yielded this:
```
...
in element 0 of attribute 'extra_protoc_files' in 'cc_nanopb_proto_compile' rule:
invalid target name
...
target names may not start with '/'
```
But it also doesn't seem to do anything, I still have to give `extra_protoc_args` the absolute path. And this is corroborate by setting `verbose` on `cc_nanopb_proto_library` to 1 or higher, the value of `extra_protoc_files` doesn't really change the parameters of the `ProtoCompile` command/call.

silvergasp, 2022-08-24:
Yeah, that's what I would expect. Have you tried it with the bazel target instead of the absolute path?

silvergasp, 2022-08-24:
The example above assumes that your options file is in the same directory as your BUILD.bazel file with the 'myproto_nanopb' target. But in theory (still untested) should work with the full Bazel target for your options file. If you are able to share the output of `tree` in your src directory I might be able to give you more specific feedback.

randyzhg, 2022-08-24:
As in setting `extra_protoc_files` then giving `extra_protoc_args = ["--nanopb_plugin_option=-f$(Bazel target)"]`? Yeah I did, it was not able to find it. Yielding either:
```
Options file not found: //bazel/subpackage/myproto.options
```
or
```
Options file not found: myproto.options
```

silvergasp, 2022-08-24:
So if your options file is at '//bazel/subpackage/myproto.options'

Then it should look something like this.
```python
cc_nanopb_proto_library(
    name = "myproto_nanopb",
    protos = [":myproto_proto"],
    extra_protoc_args = [
        "--nanopb_plugin_opt=-f$(location //bazel/subpackage/myproto.options)",
    ],
    extra_protoc_files = ["//bazel/subpackage/myproto.options"],
)
```

Note that you need the `location` prefix for the template expansion to work. [Docs here](https://bazel.build/rules/lib/ctx#expand_location).

silvergasp, 2022-08-24:
If that works we can wrap this up a little nicer, and have this handled automatically with a nanopb_options attribute.

randyzhg, 2022-08-24:
So hypothetically:
```
home
└──randyzhg
    └──myrepo
        ├──WORKSPACE.bazel
        └──subpackage
            └──proto
                ├──BUILD.bazel
                ├──myproto.proto
                └──myproto.options
```

This unfortunately did not work I get this:
```
ERROR: /home/randyzhg/myrepo/subpackage/proto/BUILD.bazel:22:24
Compiling protoc outputs for nanopb_plugin plugin on target //subpackage/proto:myproto_nanopb_pb failed:
(Exit 1): bash failed: error executing command /bin/bash -c ... (remaining 9 arguments skipped)
```

silvergasp, 2022-08-24:
Hmm, this might require some upstream changes to the grpc rules. It appears that they don't support location expansion.

randyzhg, 2022-08-24:
Bummer, does this mean there's nothing we do/fix in Nanopb to get around this right now?

silvergasp, 2022-08-24:
Yeah seems like it. I know some of the crew over on the google/pigweed team are working on something, I've sent out some feelers to see where they are at with it. But for now, it's not so simple.

tpudlik, 2022-08-25:
I'm the Pigweed person looking into it :). We've run into this problem with our bazel proto generation efforts, too (https://issuetracker.google.com/issues/234874064), but don't have a solution yet.

ankit-agarwal1999, 2022-12-05:
This patch to `rules_proto_grpc` should help
```
diff --git internal/compile.bzl internal/compile.bzl
index 0f060f8..2e7f960 100644
--- internal/compile.bzl
+++ internal/compile.bzl
@@ -39,6 +39,11 @@ proto_compile_attrs = {
         values = ["PREFIXED", "NO_PREFIX"],
         doc = "The output mode for the target. PREFIXED (the default) will output to a directory named by the target within the current package root, NO_PREFIX will output directly to the current package. Using NO_PREFIX may lead to conflicting writes",
     ),
+    "nanopb_options_file": attr.label(
+        doc = "Optional Nanopb options file for nanopb specfiically. Patched for Applied Intuition since options files couldn't be passed to Bazel",
+        allow_single_file = True,
+        mandatory = False,
+    )
 }

 def proto_compile_impl(ctx):
@@ -58,10 +63,13 @@ def proto_compile_impl(ctx):
     # Load attrs that we pass as args
     # This is done to allow writing rules that can call proto_compile with mutable attributes,
     # such as in doc_template_compile
+
     options = ctx.attr.options
     extra_protoc_args = getattr(ctx.attr, "extra_protoc_args", [])
     extra_protoc_files = ctx.files.extra_protoc_files
-
+    if ctx.file.nanopb_options_file:
+        extra_protoc_args = extra_protoc_args + ["--nanopb_plugin_opt=-f{}".format(ctx.file.nanopb_options_file.path)]
+        extra_protoc_files = ctx.files.extra_protoc_files + [ctx.file.nanopb_options_file]
     # Execute with extracted attrs
     return proto_compile(ctx, options, extra_protoc_args, extra_protoc_files)
```

PetteriAimonen, 2023-10-19:
This should now be fixed by the pull requests #904 and #906, using the option `nanopb_options_files` to set the name of the options file.

PetteriAimonen, 2023-11-11:
Fix published in nanopb-0.4.8.

closed 2023-11-11 by PetteriAimonen