nanopb/nanopb #806: Macros *_init_zero and *_init_default may produce strange initializers in C code

heikki-heikkila, 2022-08-22:
I downloaded nanopb 0.4.6 and studied nanopb/examples/network_server.  After generating fileproto.pb.h, I saw these two macros there:
```
#define ListFilesResponse_init_default           {false, false, {{NULL}, NULL}}
#define ListFilesResponse_init_zero              {false, 0, {{NULL}, NULL}}
```
These are not reasonable, as initializers for a C struct.  Namely the struct to initialize is like this:
```
typedef struct _ListFilesResponse { 
    bool has_path_error;
    bool path_error;
    DIR* file;
} ListFilesResponse;
```
So initializer should be something like ```{false,false,NULL}```.  GCC compiler tends to react to such initializers with messages like "warning: braces around scalar initializer".

heikki-heikkila, 2022-08-22:
Note that the example code (network_server/client.c) does not generate warnings because of this issue, because client.c does not use the problematic macros.  Instead it does this kind of initialization:
```
    /* Read back the response from server */
    {
        ListFilesResponse response = {};
```

PetteriAimonen, 2022-08-22:
Indeed. The generation of the initializers does not work very well with `callback_datatype`.

I'm not sure what would be a reasonable fix that would work for a wide variety of types one might specify for `callback_datatype`.

heikki-heikkila, 2022-08-23:
Perhaps you should just make a documented limitation: if callback_datatype is used, then macros *_init_zero and *_init_default must be defined by the user.

PetteriAimonen, 2022-08-23:
Perhaps, or there could be a separate `initializer` setting that can be used per field.

heikki-heikkila, 2022-08-23:
By the way, in my own project I use this kind of workaround after having #included the generated header; although my callback_datatype is a complicated struct, this workaround happens to be perfect:
```
#undef XxxMessage_init_default
#define XxxMessage_init_default {0}
```

PetteriAimonen, 2022-08-23:
Yeah, though I don't see a point to use the define in that case. You could just write `{}` or `{0}` directly, like in any C code.

PetteriAimonen, 2023-10-19:
Added some heuristics that should work for common cases, and option `initializer` for overriding in more complex cases.

PetteriAimonen, 2023-11-11:
Fix published in nanopb-0.4.8.

closed 2023-11-11 by PetteriAimonen