nanopb/nanopb #811: Invalid extras_require directive in poetry-generated setup.py

Apteryks, 2022-08-31:
Hello,

Perhaps due to a previous bug in poetry (https://github.com/python-poetry/poetry/issues/5223), the setup.py file found in the Python source archive contains an invalid extras_require directive that fails like using Python 3.9.9:

```
starting phase `build'
running "python setup.py" with command "build" and parameters ()
error in nanopb setup command: 'extras_require' must be a dictionary whose values are strings or lists of strings containing valid project/version requirement specifiers.
```

Thanks.

PetteriAimonen, 2022-08-31:
Thanks, I think this is fixed now and fix is available in pre-release `0.4.7.dev1347` on PyPI.
It can be installed with e.g.

     pip install --pre --upgrade nanopb

Apteryks, 2022-08-31:
I'm now using the poetry-based build system and building from git, to have the tests available, but thanks for fixing it promptly!

closed 2022-08-31 by Apteryks