nanopb/nanopb #812: Any details about CVE-2022-20203 reg. NanoPb 0.4.6?

lxtrasher, 2022-09-12:
Hi all,

a colleague pointed out that with [CVE-2022-20203](https://nvd.nist.gov/vuln/detail/CVE-2022-20203) there's a NanoPB issue in Android 12L. I couldn't find further details about the issue itself. I was assuming that Android has some older built-in version that suffers any vuln, but couldn't confirm it. 

Do you have any details about it? I just want to make sure that there's no issue in 0.4.6.

Thanks alot,
Alex

PetteriAimonen, 2022-09-12:
Google has not communicated with me in any way regarding this.
Looking at their [commit log](https://android.googlesource.com/platform/external/nanopb-c/+log), it appears all they have done is upgrade from 0.3.9.2 to 0.3.9.8.

Based on this the CVE-2022-20203 corresponds to combined fixes of GHSA-gcx3-7m76-287p, GHSA-3p39-mfxg-hrq4, GHSA-85rr-4rh9-hhwh, GHSA-7mv5-5mxh-qg88. The latest of these was fixed in 0.4.5.

To my best knowledge, there are currently no known vulnerabilities in 0.4.6.

lxtrasher, 2022-09-12:
Thanks for your response. Closing this.

closed 2022-09-12 by lxtrasher

tweenk, 2022-09-24:
I can confirm that CVE-2022-20203 is a vulnerability in the version of nanopb shipped with Android 12 and earlier, and does not include any new vulnerabilities in nanopb.