nanopb/nanopb #816: protoc cant decode output of proto file with 'bytes' as 'oneof' submessage

makas005, 2022-09-27:
Hello there, 
i have the following problem, which I'm not sure if it is a bug or if it is me using the library wrong.

Take the following `simple.proto` file:

```
//File: simple.proto
syntax = "proto2";

message simple {
    oneof property {
        submsg submessage = 4;
        alternative alt_message = 5;
    }
}

message submsg {
    optional bytes id = 1;
}

message alternative {
    optional uint32 alt = 1;
}
```

with the following `simple.options` file:
```
//File: simple.options
submsg.id    max_size:3
```

I do some basic setting in my program: 
```
simple s = simple_init_zero;
s.which_property = simple_submessage_tag;
s.property.submessage.has_id = false;
s.property.submessage.id.size = 1;
s.property.submessage.id.bytes[0] = 66;
```

and put out the data via:
```
uint8_t out_buffer[1024] = {0};
size_t message_length;
pb_ostream_t stream = pb_ostream_from_buffer(out_buffer, sizeof(out_buffer));
bool status = pb_encode(&stream, simple_fields, message);
message_length = stream.bytes_written;
    
if (!status)
{
    printf("Encoding failed: %s\n", PB_GET_ERROR(&stream));
}
else {
    printf("Encoded %lld bytes!", message_length);
}
FILE* outp;
outp = fopen("simple.txt", "w");
dumphex(out_buffer, message_length, outp);
fclose(outp);
outp = fopen("simple.bin", "w");
fwrite(out_buffer, sizeof(out_buffer[1]), message_length, outp);
fclose(outp);
```
(`dumphex()` just puts out the data in human readable format)

When I try to decode the output.bin file using `protoc --decode=simple  simple.proto < simple.bin` protoc will return with `Failed to parse input.`

However, tools like [https://protobuf-decoder.netlify.app/](https://protobuf-decoder.netlify.app/) will decode just fine (22030A0142 is the hex-string output).

Im happy to help or to provide any more information. Thank you

PetteriAimonen, 2022-09-27:
> outp = fopen("simple.bin", "w");

You probably want `wb` as the mode here, for binary mode. Otherwise Windows will change 0x0A to 0x0D 0x0A.

If that's not the problem, use some hex editor to verify your binary file matches your hex dump.

makas005, 2022-09-28:
`wb` was indeed the problem, thanks for your swift response!

closed 2022-09-28 by makas005