nanopb/nanopb #820: Option package prefix also msg from other .proto

edelanghe-ledger, 2022-09-30:
Hi, I'm using protobuf and nanopb in some projects and I follow the google recommendations for the directories architectures
<orga>/<project>/<packages>/<version>/<proto_files>

As this path is really long it generate struct like `<orga>_<project>_<package>_<version>_<MSG_NAME>`.
So I use the `option (nanopb_fileopt).package = "<project>_<package>"` to reduce message name to `<project>_<package>_<MSG_NAME>`

Now
I just wanted to import a proto (A) into an other (B) and use it's msg into.

A.proto
```
package <orga>/<project>/<B_packages>/<version>/<A_proto_file>
option (nanopb_fileopt).package = "<project>_<A_package>"

message Tata {  ...  }
```

B.proto
```
package <orga>/<project>/<B_packages>/<version>/<B_proto_file>
option (nanopb_fileopt).package = "<project>_<B_package>"

Import <orga>/<project>/<A_packages>/<version>/<A_proto_file>

message Toto {
   <orga>.<project>.<A_packages>.<version>.Tata tata = 1;
}
```

The issue is that the generated code for B.proto contains a `struct <project>_<B_package>_Toto` that have a field `tata` prefixed by `<project>_<B_package>` and `<orga>/<project>/<A_packages>/<version>`
Like:
```
struct <project>_<B_package>_Toto {
   <project>_<B_package>_<orga>_<project>_<A_packages>_<version>_Tata tata;
}
```

Of course no one declare this type tata. My only solution now is to NOT now use `option (nanopb_fileopt).package` and keep really long names....

The first part of the issue can be reproduced easily by adding the following into `examples/cmake_relpath/proto/simple.proto`
```
import "nanopb.proto";
option (nanopb_fileopt).package = "simple";
```
If we look at the generated simple.pb.c
```
typedef struct _simple_SimpleMessage {
    int32_t lucky_number;
    simpleUnluckyNumber unlucky;
} simple_SimpleMessage;
```

I hope this is clear. I'm doing something wrong or is it a bug ?
Thanks

PetteriAimonen, 2022-09-30:
Sounds somewhat similar to #783.

Probably a bug, but possibly there could be some workaround also.

edelanghe-ledger, 2022-10-03:
Yes, it's look like. Do you want I close this one and we continue chatting on the other ?

Btw, I check the code and I think we can fix this issue by adding a check when we append package.
```
-        if self.file_options.package:
+        if self.file_options.package and typename.startswith(self.strip_prefix):
            return "." + self.replacement_prefix + typename
```

It will force people to define the proto package to use it but I think it's for the best.
Otherwise we could add a new TypenameMangling to use this.

What do you think ?

PetteriAimonen, 2022-10-03:
I'm not sure, I have trouble understanding the details currently.. been quite overloaded lately and it is unlikely I have time to dedicate to this any time soon.

I wonder if that change could have any backward/forward compatibility for users?

edelanghe-ledger, 2022-10-03:
It's okay, take your time, I have a workaround this in my code for now :)

Yes it break backward compatibility, because if users doesn't have defined package (the one from protobuf) the `option (nanopb_fileopt).package` will stop working for them.

So we have 2 options here:
 1) We say that user must define a package because otherwise we cannot detect if the field `.sub.UnluckyNumber` is a `msg UnluckyNumber` from a `package sub` or a nested `msg UnluckyNumber` into a `msg sub`.
 2) We don't want to break compatibility for users and we add a new mangling option, that does the startswith check.
 
 WDYT ?

PetteriAimonen, 2022-10-03:
Breaking compatibility is something I try to avoid unless absolutely necessary.

Third possible approach could be to generate some `#define` directives that would always allow access to the types by long names, even if the generated header uses short names.

PetteriAimonen, 2022-11-24:
Should be fixed now. The `.pb.h` file gets a list of defines to map the files to their original/canonical names, for use by other files.

closed 2022-12-11 by PetteriAimonen