nanopb/nanopb #821: oneof field decodes buffer wrong

iCenger, 2022-10-05:
Hi,
First of all this is my header file. 
[messages.pb.zip](https://github.com/nanopb/nanopb/files/9713552/messages.pb.zip)
I have a problem with decoding. 
Here is my structure and its location.
[image]
And datas in the memory.
[image]

As you can see, value at 0x2001eecc and value of transaction_id is same.
Also value at 0x2001eed0 and value of which_payload is same. Problem starts after that. 

challenge_code's location is 0x2001eed2, i think its correct(because which_payload's size pb_size_t which means 16bit and 0x2001eed0 + 2 = 0x2001eed2). But when decoding, datas are starting to written 0x2001eed4. You can see memory map. So it gets value wrong(It should be 0x000bc2bf instead of 0xc2bf0000). Following variables(token and email) also wrong.

I hope I explain my problem correctly. What causes this? Am I missing something?

Regards.

PetteriAimonen, 2022-10-05:
It seems you may have different compilation settings in different compilation units.
That is causing compiler to layout the structure differently in memory.

Are you using only C, or both C and C++ in this project?
What compiler and what compiler flags?

iCenger, 2022-10-05:
I did clean and full rebuild. I'm using only C, in IAR Embedded WorkBench 9.30 and there are no optimizations. I'm cleaning and rebuilding every single time.

PetteriAimonen, 2022-10-05:
Try putting this debug code in both `messages.pb.c` and in your own code where you try to access the data.

By editing the autogenerated `messages.pb.c`:

    const int DEBUG_payload_pair_A = offsetof(Request, payload.pair);

In your own code outside of any functions:

    const int DEBUG_payload_pair_B = offsetof(Request, payload.pair);

Then check if the two values match in debugger.
They should be either 6 or 8, depending on whether the compiler adds padding.

(Note that the editing of messages.pb.c will trigger a recompile, so if for some reason the clean is not really cleaning messages.pb.o and you have had some change in compiler settings, that could magically make the bug disappear.)

iCenger, 2022-10-05:
Thanks! 

#pragma pack(1)
typedef struct _Request { 
    uint32_t transaction_id;
    pb_size_t which_payload;
    union {
        PairReadyRequest pairReady;
        PairRequest pair;
        TrackRequest track;
    } payload;
} Request;

I simply add #pragma pack in my code and bug fixed.

PetteriAimonen, 2022-10-05:
You can do that also by specifying `options (nanopb_fileopt).packed_struct = true;`.

But I really suggest you figure out why you are otherwise getting different struct layout in different compilation units. Otherwise you'll just hit the same bug with some other struct.

iCenger, 2022-10-05:
Thanks, i will check that.

closed 2022-10-05 by iCenger