nanopb/nanopb #824: Nanopb encoding error

iCenger, 2022-10-18:
Hi Petteri,

I have a issue with encoding.

**_First of all here is my .proto :_**

message AutoDiscoverySlave {
  uint32 address = 1;
  string model = 2;
}

message AutoDiscoveryStatus{
  uint32 progress = 1;
  repeated AutoDiscoverySlave slaves = 2;
}

**_My .option file_**

AutoDiscoverySlave.model   max_length:20
AutoDiscoveryStatus.slaves max_count:5

**_And there is generated structure_**

typedef struct _AutoDiscoverySlave { 
    uint32_t address;
    char model[21];
} AutoDiscoverySlave;

typedef struct _AutoDiscoveryStatus { 
    uint32_t progress;
    pb_size_t slaves_count;
    AutoDiscoverySlave slaves[5];
} AutoDiscoveryStatus;

Encoding was working perfectly  and when its working stream.bytes_written was 66.

After that, i did some refactoring but didnt change .proto, .options, .pb.h or encoding part. Now when i tried to encode 
stream.bytes_written = 2 and stream.errmsg = 0xFFFFFF, no error i guess because pb_encode returns 1.

**_Here is my encode part :_** 

void enda_encode_publish_data(const void *buf, uint32_t transaction_id,
                              uint16_t len, enda_publish_type_t publish_type)
{
const AutoDiscoveryStatus *auto_discovery = (const AutoDiscoveryStatus *)buf;

stream = pb_ostream_from_buffer(mqtt_queue_buf.publish_data, sizeof(mqtt_queue_buf.publish_data));
status = pb_encode(&stream, AutoDiscoveryStatus_fields, auto_discovery);
message_len = stream.bytes_written;
}
Seems like, it just encodes
    uint32_t progress;
    pb_size_t slaves_count;
these two.

Im trying to encode same datas in same way. What am I missing? Could you help ?

Regards

Ibrahim

PetteriAimonen, 2022-10-18:
What was the root cause of your problem #821?

It could be causing this one also, and until you figure out your compiler problems, I don't find this worth my time.

iCenger, 2022-10-18:
That one was, i used #pragma pack 
(1) and did not close it with #pragma pack(). So that can not cause this. Thanks for the reply

closed 2022-10-18 by iCenger