nanopb/nanopb #826: encode error

shaoxiaoxi, 2022-10-21:
Hi,
I have a issue with encoding.

First of all here is my .proto 
syntax = "proto2";
message StudentName {
    required string f_name = 1;
    required string l_name = 2;
}

message Student
{
    required uint32 num        = 1;
    required uint32 age        = 2;
    optional string c_score    = 3;
    optional StudentName sname = 4;
}

My .option file

StudentName.f_name max_size:16
StudentName.l_name max_size:16
Student.c_score max_size:16
Student.sname max_count:5

And there is generated structure
/* Struct definitions */
typedef struct _StudentName {
    char f_name[16];
    char l_name[16];
/* @@protoc_insertion_point(struct:StudentName) */
} StudentName;

typedef struct _Student {
    uint32_t num;
    uint32_t age;
    bool has_c_score;
    char c_score[16];
    bool has_sname;
    StudentName sname;
/* @@protoc_insertion_point(struct:Student) */
} Student;

The encoding works perfectly, when has_sname is 0, pb_encode fails when has_sname is 1.

It is suspected that iter does not point to the PB_LAST_FIELD of Student_fields after processing StudentName_fields.

What am I missing? Could you help ?

Regards

Ailenda

PetteriAimonen, 2022-10-21:
Which version of nanopb are you using?

> The encoding works perfectly, when has_sname is 0, pb_encode fails when has_sname is 1.

How does it fail? If it returns `false`, what is the error message in `stream.errmsg`?

shaoxiaoxi, 2022-10-21:
The version is 0.3.9.6, the same problem occurs with 0.4.6.
stream.errmsg is none

PetteriAimonen, 2022-10-21:
So, is `pb_encode()` returning false? Does it write any bytes to the output stream?

shaoxiaoxi, 2022-10-22:
Yes, pb_encode() returning false. There are some  bytes in the output stream.

When processing tag 2 of StudentName_fields, the order of iter->pos->tag and prev_field->tag obtained by the function pb_field_iter_next() is:
iter->pos->tag=0; prev_field->tag=2；
iter->pos->tag=0; prev_field->tag=2；
iter->pos->tag=0; prev_field->tag=2；
iter->pos->tag=0; prev_field->tag=2；
bool pb_field_iter_next(pb_field_iter_t *iter)
{
    const pb_field_t *prev_field = iter->pos;
    if (prev_field->tag == 0)  {
        return false;
    }
    iter->pos++;

    if (iter->pos->tag == 0)
   ...................
}

PetteriAimonen, 2022-11-24:
I'm currently unable to reproduce this problem.

closed 2022-11-24 by PetteriAimonen

shaoxiaoxi, 2022-11-24:
> I'm currently unable to reproduce this problem.

This issue occurs on the keil C51 system, and the multi-level call on the C51 will be abnormal. Problem has been resolved.