nanopb/nanopb #827: Error with max_size with and max_count in options file

requinham, 2022-10-24:
Hello, I want to share with you a compilation error when I define max_size and max_count options for a "repeated bytes" or "repeated string"
for example:
.proto file:
`
message Signature {
int32 signatureId = 1;
repeated bytes certificateChain = 2;
}
`
.options file:
`Signature.certificateChain           max_size:1200 max_count:4`
proto generation pass without problem but when compiling I got this kind of error below (in end of this post).

BUT, when I put this options in .proto file like this 
`
message Signature {
int32 signatureId = 1;
repeated bytes certificateChain = 2      [(nanopb).max_size = 1200, (nanopb).max_count = 4]; 
}
`
It compile without problem!!!

ERROR:
`error: static assertion failed: "FIELDINFO_DOES_NOT_FIT_width4_field6"
  183 | #      define PB_STATIC_ASSERT(COND,MSG) _Static_assert(COND,#MSG);
      |                                          ^~~~~~~~~~~~~~
lib/nanopb/pb.h:788:5: note: in expansion of macro ‘PB_STATIC_ASSERT’
  788 |     PB_STATIC_ASSERT(PB_FITS(tag,16) && PB_FITS(data_offset,16) && PB_FITS((int_least8_t)size_offset,8) && PB_FITS(data_size,16) && PB_FITS(array_size,16), FIELDINFO_DOES_NOT_FIT_width4_field ## tag)
      |     ^~~~~~~~~~~~~~~~
lib/nanopb/pb.h:587:5: note: in expansion of macro ‘PB_FIELDINFO_ASSERT_4’
  587 |     PB_FIELDINFO_ASSERT_4(tag, PB_ATYPE_ ## atype | PB_HTYPE_ ## htype | PB_LTYPE_MAP_ ## ltype, \
      |     ^~~~~~~~~~~~~~~~~~~~~
note: in expansion of macro ‘PB_GEN_FIELD_INFO_ASSERT_4’
  335 | X(a, STATIC,   SINGULAR, BYTES,    apiVersionId,      6) \
      | ^
lib/nanopb/pb.h:518:5: note: in expansion of macro ‘com_fca_uconnect_global_UconnectMessage_FIELDLIST’
  518 |     msgname ## _FIELDLIST(PB_GEN_FIELD_INFO_ASSERT_ ## width, structname)
      |     ^~~~~~~
note: in expansion of macro ‘PB_BIND’
    9 | PB_BIND(com_fca_uconnect_global_UconnectMessage, com_fca_uconnect_global_UconnectMessage, 4)
      | ^~~~~~~
lib/nanopb/pb.h:183:42: error: static assertion failed: "FIELDINFO_DOES_NOT_FIT_width4_field7"
  183 | #      define PB_STATIC_ASSERT(COND,MSG) _Static_assert(COND,#MSG);
      |                                          ^~~~~~~~~~~~~~
lib/nanopb/pb.h:788:5: note: in expansion of macro ‘PB_STATIC_ASSERT’
  788 |     PB_STATIC_ASSERT(PB_FITS(tag,16) && PB_FITS(data_offset,16) && PB_FITS((int_least8_t)size_offset,8) && PB_FITS(data_size,16) && PB_FITS(array_size,16), FIELDINFO_DOES_NOT_FIT_width4_field ## tag)
      |     ^~~~~~~~~~~~~~~~
lib/nanopb/pb.h:587:5: note: in expansion of macro ‘PB_FIELDINFO_ASSERT_4’
  587 |     PB_FIELDINFO_ASSERT_4(tag, PB_ATYPE_ ## atype | PB_HTYPE_ ## htype | PB_LTYPE_MAP_ ## ltype, \
      |     ^~~~~~~~~~~~~~~~~~~~~
lib/stellantis/Global_Uconnect_Control.pb.h:336:1: note: in expansion of macro ‘PB_GEN_FIELD_INFO_ASSERT_4’
  336 | X(a, STATIC,   OPTIONAL, MESSAGE,  messageSignature,   7)
      | ^
lib/nanopb/pb.h:518:5: note: in expansion of macro ‘com_fca_uconnect_global_UconnectMessage_FIELDLIST’
  518 |     msgname ## _FIELDLIST(PB_GEN_FIELD_INFO_ASSERT_ ## width, structname)
      |     ^~~~~~~
:9:1: note: in expansion of macro ‘PB_BIND’
    9 | PB_BIND(com_fca_uconnect_global_UconnectMessage, com_fca_uconnect_global_UconnectMessage, 4)
      | ^~~~~~~
lib/nanopb/pb.h:183:42: error: static assertion failed: "FIELDINFO_DOES_NOT_FIT_width4_field2"
  183 | #      define PB_STATIC_ASSERT(COND,MSG) _Static_assert(COND,#MSG);
      |                                          ^~~~~~~~~~~~~~
lib/nanopb/pb.h:788:5: note: in expansion of macro ‘PB_STATIC_ASSERT’
  788 |     PB_STATIC_ASSERT(PB_FITS(tag,16) && PB_FITS(data_offset,16) && PB_FITS((int_least8_t)size_offset,8) && PB_FITS(data_size,16) && PB_FITS(array_size,16), FIELDINFO_DOES_NOT_FIT_width4_field ## tag)
      |     ^~~~~~~~~~~~~~~~
lib/nanopb/pb.h:587:5: note: in expansion of macro ‘PB_FIELDINFO_ASSERT_4’
  587 |     PB_FIELDINFO_ASSERT_4(tag, PB_ATYPE_ ## atype | PB_HTYPE_ ## htype | PB_LTYPE_MAP_ ## ltype, \
      |     ^~~~~~~~~~~~~~~~~~~~~
lib/stellantis/Global_Uconnect_Control.pb.h:349:1: note: in expansion of macro ‘PB_GEN_FIELD_INFO_ASSERT_4’
  349 | X(a, STATIC,   OPTIONAL, MESSAGE,  V2CMessage,        2) \
      | ^
lib/nanopb/pb.h:518:5: note: in expansion of macro ‘com_fca_uconnect_global_UnsignedDataMsg_FIELDLIST’
  518 |     msgname ## _FIELDLIST(PB_GEN_FIELD_INFO_ASSERT_ ## width, structname)
      |     ^~~~~~~
:15:1: note: in expansion of macro ‘PB_BIND’
   15 | PB_BIND(com_fca_uconnect_global_UnsignedDataMsg, com_fca_uconnect_global_UnsignedDataMsg, 4)
      | ^~~~~~~
lib/nanopb/pb.h:183:42: error: static assertion failed: "FIELDINFO_DOES_NOT_FIT_width4_field3"
  183 | #      define PB_STATIC_ASSERT(COND,MSG) _Static_assert(COND,#MSG);
      |                                          ^~~~~~~~~~~~~~
lib/nanopb/pb.h:788:5: note: in expansion of macro ‘PB_STATIC_ASSERT’
  788 |     PB_STATIC_ASSERT(PB_FITS(tag,16) && PB_FITS(data_offset,16) && PB_FITS((int_least8_t)size_offset,8) && PB_FITS(data_size,16) && PB_FITS(array_size,16), FIELDINFO_DOES_NOT_FIT_width4_field ## tag)
      |     ^~~~~~~~~~~~~~~~
lib/nanopb/pb.h:587:5: note: in expansion of macro ‘PB_FIELDINFO_ASSERT_4’
  587 |     PB_FIELDINFO_ASSERT_4(tag, PB_ATYPE_ ## atype | PB_HTYPE_ ## htype | PB_LTYPE_MAP_ ## ltype, \
      |     ^~~~~~~~~~~~~~~~~~~~~
lib/stellantis/Global_Uconnect_Control.pb.h:350:1: note: in expansion of macro ‘PB_GEN_FIELD_INFO_ASSERT_4’
  350 | X(a, STATIC,   OPTIONAL, MESSAGE,  C2VMessage,        3) \
      | ^
lib/nanopb/pb.h:518:5: note: in expansion of macro ‘com_fca_uconnect_global_UnsignedDataMsg_FIELDLIST’
  518 |     msgname ## _FIELDLIST(PB_GEN_FIELD_INFO_ASSERT_ ## width, structname)
      |     ^~~~~~~
:15:1: note: in expansion of macro ‘PB_BIND’
   15 | PB_BIND(com_fca_uconnect_global_UnsignedDataMsg, com_fca_uconnect_global_UnsignedDataMsg, 4)
      | ^~~~~~~
lib/nanopb/pb.h:183:42: error: static assertion failed: "FIELDINFO_DOES_NOT_FIT_width4_field4"
  183 | #      define PB_STATIC_ASSERT(COND,MSG) _Static_assert(COND,#MSG);
      |                                          ^~~~~~~~~~~~~~
lib/nanopb/pb.h:788:5: note: in expansion of macro ‘PB_STATIC_ASSERT’
  788 |     PB_STATIC_ASSERT(PB_FITS(tag,16) && PB_FITS(data_offset,16) && PB_FITS((int_least8_t)size_offset,8) && PB_FITS(data_size,16) && PB_FITS(array_size,16), FIELDINFO_DOES_NOT_FIT_width4_field ## tag)
      |     ^~~~~~~~~~~~~~~~
lib/nanopb/pb.h:587:5: note: in expansion of macro ‘PB_FIELDINFO_ASSERT_4’
  587 |     PB_FIELDINFO_ASSERT_4(tag, PB_ATYPE_ ## atype | PB_HTYPE_ ## htype | PB_LTYPE_MAP_ ## ltype, \
      |     ^~~~~~~~~~~~~~~~~~~~~
lib/stellantis/Global_Uconnect_Control.pb.h:351:1: note: in expansion of macro ‘PB_GEN_FIELD_INFO_ASSERT_4’
  351 | X(a, STATIC,   OPTIONAL, MESSAGE,  session,           4)
      | ^
lib/nanopb/pb.h:518:5: note: in expansion of macro ‘com_fca_uconnect_global_UnsignedDataMsg_FIELDLIST’
  518 |     msgname ## _FIELDLIST(PB_GEN_FIELD_INFO_ASSERT_ ## width, structname)
      |     ^~~~~~~
:15:1: note: in expansion of macro ‘PB_BIND’
   15 | PB_BIND(com_fca_uconnect_global_UnsignedDataMsg, com_fca_uconnect_global_UnsignedDataMsg, 4)
      | ^~~~~~~
make: *** [Makefile:4: dk_reader] Error 1
`

PetteriAimonen, 2022-10-24:
Do you have only a single `.proto` file, or several `.proto` files that import each other?

It's likely that there is some problem finding the `.options` file associated with the imported `.proto` file for some reason. This would then cause not enough space to be allocated for the message descriptors.

To actually be able to debug and fix this, I would need a set of proto files and the compilation commands that reproduce the problem.

requinham, 2022-10-24:
@PetteriAimonen I update the bug description. Please take a look on it and tell me if you need other information.

PetteriAimonen, 2022-10-24:
I don't see the necessary files and commands to reproduce this problem.

requinham, 2022-10-24:
Please find example attached.
command:
`$HOME/nanopb/generator/nanopb_generator.py -I . --output-dir=. ./test.proto ./test_ext.proto`
`make clean; make`
It seems that a total message limit blocks the compilation.
[Archive.zip](https://github.com/nanopb/nanopb/files/9850308/Archive.zip)

PetteriAimonen, 2022-10-24:
Thanks, I was able to reproduce the problem now.

PetteriAimonen, 2022-10-24:
Ok, it appears the build error is actually due to `PB_FIELD_32BIT` not being defined. If you add `-DPB_FIELD_32BIT` it builds ok. As documented [here](https://jpa.kapsi.fi/nanopb/docs/reference.html#compilation-options) that flag is needed for fields larger than 64 kB. The 8kB * 100 = 800 kB exceeds the limit.

I'll try to add a better error message for this.

requinham, 2022-10-24:
Thank you so much. Indeed, the error message is not clear. It is clear and compile without error now.

closed 2022-10-24 by requinham