nanopb/nanopb #828: Error generating protobuf with an enum alias

dlizewski, 2022-10-31:
When generating a protobuf that has an emun alias.

Example:
syntax = "proto3";
enum SomethingWithAlias {
    option allow_alias = true;
    item0 = 0;
    item1 = **1**;
    item1_alternate = **1**;
    item2 = 2;
}

The function auxiliary_defines uses "sorted_values = sorted(self.values, key = lambda x: (x[1], x[0]))" to sort the keys by value.

I assume x[0] was to have a secondary sort by the name, but **x[0] is an object (class Names) that is not comparable**.
Running "python3 nanopb/generator/nanopb_generator.py nanopb/test_alias.proto"

results in: 
File "nanopb/generator/nanopb_generator.py", line 268, in auxiliary_defines
    sorted_values = sorted(self.values, key = lambda x: (x[1], x[0]))
TypeError: '<' not supported between instances of 'Names' and 'Names'

To fix this, the secondary sort should point to the name of the enum using x[0].parts[0]

sorted_values = sorted(self.values, key = lambda x: (x[1], x[0]**.parts[0]**))

dlizewski, 2022-10-31:
or better yet, implement the __lt__ and __gt__ methods for class Names

PetteriAimonen, 2022-10-31:
I'm not getting that error when compiling the `.proto` file you included with any of the versions I tried (0.3.9, 0.4.5, newest git).

Do you have something else in the proto file also?

PetteriAimonen, 2022-10-31:
This seems like a duplicate of bug #535, fixed in 2020 in version 0.4.2.

dlizewski, 2022-10-31:
Sorry, you are right, looks like my org was using 0.4.1. Closing, thanks for responding

closed 2022-10-31 by dlizewski