nanopb/nanopb #829: "Parent Stream too short" when message size exceeds 100 bytes

stormyweathers, 2022-11-10:
Hey there,

I am decoding my messages by calling `pb_istream_from_buffer` and then `pb_decode` after each `char` is received.   This works fine for small messages, but when the message size exceeds 100 bytes, `pb_decode` throws the error "parent stream too short".

I haven't been able to understand the documentation or source code to figure out exactly what is going on.  Something to do with substreams that is over my head.

Do you have any advice? 

This is my message receiving code:
`
void receive_message(pb_byte_t stream_buffer[],const pb_field_t message_fields[], msgType *message_ptr) {
  size_t bytes_read = 0;
  while (1) {
    if (pc_.readable()) {

      // Create a stream that reads from the buffer.
      char c = pc_.getc();
      stream_buffer[bytes_read] = c;
      bytes_read++;

      // Try decoding every time
      pb_istream_t stream = pb_istream_from_buffer(
          static_cast<pb_byte_t *>(stream_buffer), bytes_read);

      // Now we are ready to decode the message.
      bool status =
          pb_decode(&stream, message_fields, message_ptr);

      // Check for errors...
      if (status) {
        // Decode succeeded. Message is now ready for use.
        return;
      }

      // Take another character and try again.
    }
  }
}
`

PetteriAimonen, 2022-11-10:
What is the actual message data you are trying to decode, e.g. as a hex dump?

If you are on Windows, make also sure that the `pc_` stream is set to binary mode, to avoid Windows from altering 0x0A byte values.

But in reality, a protobuf message may decode successfully even before it has been fully received, so I would recommend more explicit way to pass the message length. Such as `pb_encode_delimited()`.

stormyweathers, 2022-11-10:
Thanks for the timely reply.

I am running ubuntu, and here is the hex dump I get from `print(msg.SerializeToString())`: 
`b'\n\x1a\x08\x00\x10\x00\x18\x00 \x00(\x000\x008\x8b\x1a@\x9f\x10H\x00P\x00X\x00`\x00\x12\x18\x08\xeb\x03\x10\xff\x1f\x18\x99\x13 \x8e\x11(\x000\x008\x00@\x00H\x00P\x01\x1a\x1a\x08\x00\x10\x00\x18\x00 \x00(\x000\x008\x8b\x1a@\x9f\x10H\x00P\x00X\x00`\x00"\x18\x08\xeb\x03\x10\xff\x1f\x18\x99\x13 \x8e\x11(\x000\x008\x00@\x00H\x00P\x01'`

Ok I will look into `pb_encode_delimited()`.  In the case a message is decoded before fully received, I suppose I should flush the serial buffer to avoid getting the leftover bytes on the next message.

PetteriAimonen, 2022-11-10:
That message data appears corrupted, for example `protoc --decode_raw` does not accept it.

stormyweathers, 2022-11-10:
> Support for delimiting varies a between between libraries, e.g.
> https://github.com/protocolbuffers/protobuf/blob/main/src/google/protobuf/util/delimited_message_util.h
> has it for C++. I think the Python protobuf library still doesn't have
> support built-in, though e.g. https://pypi.org/project/delimited-protobuf/
> provides it. The general idea is mentioned in docs here:
> https://developers.google.com/protocol-buffers/docs/techniques#streaming
>
> Actually the data seems to be accepted by protoc --decode_raw fine now.
> Before your edit the Github formatting was just eating out the ` so I got a
> corrupted 106 byte message instead of the correct 108 bytes.
>
> You may want to verify the data you are actually passing to pb_decode()
> also, to make sure there is no further corruption in between. Especially
> serial ports are prone to eating some byte values if they are not correctly
> set to binary mode.

I tested the data integrity by passing back each char received over the
same `pc_` serial port and inspecting on the python side.  It all checks
out.

Can you elaborate what you mean by correctly setting to binary mode?

PetteriAimonen, 2022-11-10:
Use a debugger and check the buffer before you call `pb_decode`, and also check the size you pass to `pb_istream_from_buffer()`.

stormyweathers, 2022-11-10:
Ok i will try that.
Ive been working on an stm32 microcontroller, without using a debugger.
Ive been printing over the serial connection or an SPI OLED display for
debugging. I will first check the buffer using these methods.

 it might take me a bit to figure out how to use the debugger.

PetteriAimonen, 2022-11-10:
STM32 has pretty good debugger support, and in my experience it is easy enough to set up also. A cheap STM32 discovery board or similar is enough as the debugger adapter.

Even if the data is correct, the next step will be to start stepping into the code to see what is actually happening, so you'll need some way to run a debugger anyway. Often a reasonable alternative is also to compile your code for PC instead and debug there first.

stormyweathers, 2022-11-10:
I'm using an stm32f746zg nucleo board with a built-in st-link adapter.
But its taking me quite some time to understand how to set up the debugger.

Do you have any tutorial to suggest?
I'm still searching for one

stormyweathers, 2022-11-10:
Ok I have `open-ocd` --> `arm-none-eabi-gdb` --> `ddd` working now.  Time to start debugging

stormyweathers, 2022-11-11:
Ok I have some results.

I just went for it in raw command-line `gdb` because I couldnt get any GUI's to work. 
Each time the `    if (pc_.readable()) {` loop is run, the correct char is read from the serial line.  However, the data is not properly stored in `stream_buffer`.
For some reason, `stream_buffer` contains `\n\032\b`, and doesnt accumulate any more bytes, even when more than 3 bytes have been read from the stream.  
Furthermore, the actual message starts with `\n\x1a\x08`.
So something is going wrong with the line `stream_buffer[bytes_read] = c;`

For reference, I am calling `receive_message` as `receive_message(buffer_MOTDrop,minig_MOTDrop_fields,&MOTDrop);`
where the buffer is declared `pb_byte_t buffer_MOTDrop[minig_MOTDrop_size];`.  According to my `.proto`, `minig_MOTDrop_size=256`, so the buffer size should not be an issue.

Any ideas?

Thanks again for your help

PetteriAimonen, 2022-11-12:
If you are allocating the buffers on stack, make sure your stack is large enough to hold them.
Or you can make the buffers `static` instead, to allocate them in `.bss` section instead.

You can also debug instruction-by-instruction to see more clearly what is happening. In gdb, set `display /i $pc` to show current instruction and use `stepi` to step instruction-by-instruction (pressing enter will repeat it), `info registers` will show values in all registers.

PetteriAimonen, 2022-11-24:
I'm currently unable to reproduce this problem.

closed 2022-11-24 by PetteriAimonen