nanopb/nanopb #830: How to define PB_CONVERT_DOUBLE_FLOAT

TorpidFurball, 2022-11-11:
Hello, 

I am working with Nanopb for decoding a data stream on an Arduino Mega. The error I am getting is :

`C:\Users\jonas\Documents\Arduino\libraries\Nanopb/pb.h:180:42: error: static assertion failed: DOUBLE_MUST_BE_8_BYTES
 #      define PB_STATIC_ASSERT(COND,MSG) static_assert(COND,#MSG);
                                          ^
C:\Users\jonas\Documents\Arduino\Read_Ethernet\proto_files\egm.pb.c:82:1: note: in expansion of macro 'PB_STATIC_ASSERT'
 PB_STATIC_ASSERT(sizeof(double) == 8, DOUBLE_MUST_BE_8_BYTES)
`

I can see in the compiled .c file the instruction saying:

`/* On some platforms (such as AVR), double is really float.
 * To be able to encode/decode double on these platforms, you need.
 * to define PB_CONVERT_DOUBLE_FLOAT in pb.h or compiler command line.
 */`

I am assuming that since Arduino uses float instead of double, this is the cause of the issue and I need to define PB_CONVERT_DOUBLE_FLOAT. I have the ph.h file available to edit, but I don't know exactly how to define that variable.

Any help would be greatly appreciated.

PetteriAimonen, 2022-11-12:
Uncomment the `#define` near line 45 of `pb.h`:

```
/* Enable conversion of doubles to floats for platforms that do not
 * support 64-bit doubles. Most commonly AVR. */
/* #define PB_CONVERT_DOUBLE_FLOAT 1 */
```

TorpidFurball, 2022-11-15:
Thank you for your reply. I tried your fix above, but was still getting the Static Assert failed error in my first message. I made some more changes to the code to no avail and finally got to a place where I decided I should just reset to the original nanopb files.  

Here is some background on what I am trying to do in this project- I want the position coordinates of a robot arm which are output by the robot controller. The controller uses Google Protocol Buffer to encode the data and sends the packet over Ethernet (using UDP) to the Arduino Mega. I've attached the relevant code below. I used the exact .proto files that the robot controller uses to compile the necessary .pb.h and .pb.c files.
```
#include "proto_files/egm.pb.h"
#include <pb_decode.h>
#include <pb.h>
#include <pb_common.h>
#include <Ethernet.h>
#include <EthernetUdp.h>
#define UDP_TX_PACKET_MAX_SIZE 200

abb_egm_EgmRobot msg = abb_egm_EgmRobot_init_zero;  
char packetBuffer[UDP_TX_PACKET_MAX_SIZE];  // buffer to hold incoming packet,
EthernetUDP Udp;

void setup() {

  Ethernet.begin(mac, ipShield); //begin Ethernet
  Serial.begin(9600);
  Udp.begin(localPort); // start UDP
}

void loop() {
  int packetSize = Udp.parsePacket();// if there's data available, read a packet
  if (packetSize) {
    Serial.print("Received packet of size ");
    Serial.println(packetSize);
    }

    // read the packet into packetBufffer
    Udp.read(packetBuffer, UDP_TX_PACKET_MAX_SIZE);
    pb_istream_t stream = pb_istream_from_buffer(packetBuffer, sizeof(packetBuffer));
    bool status = pb_decode(&stream, abb_egm_EgmRobot_fields, &msg);
    if (!status){
      Serial.println("Failed to decode");
    return;
    }  
    Serial.println("Contents:");
    Serial.println(packetBuffer);
  }
}
```

I have three main questions:

1) The current error I am getting when compiling my Arduino code reads 
```
C:\Users\jonas\Documents\Arduino\libraries\Nanopb/pb_decode.c:1181: undefined reference to `abb_egm_EgmRobot_msg'
collect2.exe: error: ld returned 1 exit status
```
This references a line in the pb_decode.c file but I'm not sure what is causing the error.

2) Even after I defined the PB_CONVERT_DOUBLE_FLOAT, I was still getting the same error. I know that since my Arduino Mega does not support the 64-bit doubles so I needed to define it, but is there anywhere else that I need to define something? I also saw there is the option to disable the Static Assert check if needed. Is this something that you would recommend if this error persists?

3) I guess this is a more general question, but this program uses the Ethernet Library to read the UPD encoded data packets from the controller. The way the library is written requires that the packet be read into a "char" type variable. Does this affect nanopb? Does nanopb expect the uint8_t type?  

Thank you again.

PetteriAimonen, 2022-11-15:
`char` and `uint8_t` are equivalent for this purpose. You may need to cast the pointer but that is fine.

Regarding your other errors, I think you are probably compiling different files than you are modifying.
I wouldn't recommend disabling the check, because it won't work without enabling the conversion (that's why the check is there).

The undefined reference is probably due to the `egm.pb.c` file not being included in project for some reason.

closed 2022-11-24 by PetteriAimonen