nanopb/nanopb #833: Referencing a header file

BenWhite-Price, 2022-11-17:
I'm trying to create a string of a known length using the max_size option as outlined in [this](guide). I'm able to use a number in the code, but I'd really like to use a value defined in a separate header file.

For example, instead of 
`string email = 1 [(nanopb).max_size = 64];`
I'd like something like 
``string email = 1 [(nanopb).max_size = EMAIL_LENGTH];`
where `EMAIL_LENGTH` is a `#define`d value in a separate .h file. Is such a thing possible?

PetteriAimonen, 2022-11-18:
Currently the generator does not have support for generating such code, and it would be somewhat difficult to add.

It would work if the generated `.pb.h` file was edited to contain reference to such a symbol, though when crossing thresholds of 256 / 65536 bytes there would be further issues.

BenWhite-Price, 2022-11-18:
> Wouldn't it make more sense to have a .options.in file that is preprocessed before the generator is called ?

That sounds very interesting. Is there documentation about the .options file?

smorin-primero, 2022-11-18:
[Defining the options in a .options file](https://jpa.kapsi.fi/nanopb/docs/reference.html#defining-the-options-in-a-.options-file)

PetteriAimonen, 2022-11-24:
Closing as duplicate of #707.

closed 2022-11-24 by PetteriAimonen