nanopb/nanopb #841: Cannot import 'TemporaryDirectory'

shahsahi, 2022-12-15:
I am currently trying to migrate from nanopb 0.4.5 to 0.4.7 and running into the following issue:

```
Traceback (most recent call last):
  File "...nanopb_generator.py", line 55, in <module>
    from proto import TemporaryDirectory
ImportError: cannot import name 'TemporaryDirectory' from 'proto' (unknown location)
```
For context, the way I am using nanopb is through a Conan package, and I am calling the generator directly with python (python nanopb_generator.py). The problem seems to be that the TemporaryDirectory import moved to `__init__.py`, which is not installed in CMakeLists.txt (the package install only has the following in the proto folder):

```
nanopb
└── lib
    └── site-packages
        └── proto
            ├── nanopb_pb2.py
            ├── nanopb.proto
            └── _utils.py
```

If I copy `__init.py__` here then the generator runs without issue. I can work around this but would it make sense for this file be installed alongside `_utils.py` in CMakeLists.txt?

PetteriAimonen, 2022-12-16:
I agree. I added the missing file to CMakeLists.txt now.
Somewhat related to #556.

It would be great to have an automatic test for cmake install or conan package.
Similar to bazel, platformio, ios_swift and pypi_publish workflows here:
https://github.com/nanopb/nanopb/tree/master/.github/workflows

There is already `examples/conan_dependency`.
I wonder if just adding a workflow that builds it according to the README would be enough?

shahsahi, 2022-12-16:
This fixes my issue, thanks! And yes seems like a workflow according to the README might make sense as well.

PetteriAimonen, 2023-11-11:
The main issue should be fixed in 0.4.8. I added separate issue #922 about adding a continuous integration test for the Conan build.

closed 2023-11-11 by PetteriAimonen