nanopb/nanopb #845: The Criterion project is broken after nanopb upgrade to 0.4.7

yurivict, 2022-12-23:
https://github.com/Snaipe/Criterion

```
===>  Building for criterion-2.4.1
[  1% 1/51] /usr/ports/devel/criterion/work/Criterion-2.4.1/src/protocol/gen-pb.py /usr/local/bin/protoc /usr/local/bin/protoc-gen-nanopb ../src/protocol/criterion.proto ../src/protocol/criterion.options /usr/ports/devel/criterion/work/Criterion-2.4.1/_build/src
FAILED: src/criterion.pb.c src/criterion.pb.h 
/usr/ports/devel/criterion/work/Criterion-2.4.1/src/protocol/gen-pb.py /usr/local/bin/protoc /usr/local/bin/protoc-gen-nanopb ../src/protocol/criterion.proto ../src/protocol/criterion.options /usr/ports/devel/criterion/work/Criterion-2.4.1/_build/src
/usr/local/bin/protoc --plugin=protoc-gen-nanopb=/usr/local/bin/protoc-gen-nanopb -I../src/protocol --nanopb_out=-f ../src/protocol/criterion.options:/usr/ports/devel/criterion/work/Criterion-2.4.1/_build/src ../src/protocol/criterion.proto
Traceback (most recent call last):
  File "/usr/local/bin/protoc-gen-nanopb", line 7, in <module>
    from nanopb_generator import *
  File "/usr/local/bin/nanopb_generator.py", line 55, in <module>
    from proto import TemporaryDirectory
ImportError: cannot import name 'TemporaryDirectory' from 'proto' (unknown location)
--nanopb_out: protoc-gen-nanopb: Plugin failed with status code 1.
ninja: build stopped: subcommand failed.
```

PetteriAimonen, 2022-12-23:
Seems like duplicate of #841. I'll release 0.4.8 in a few weeks to fix that, once any other breakages have been discovered.

paolostivanin, 2023-01-30:
Hello!
The fix for #841 helps only a bit. Now we get:

```
[    9s] Traceback (most recent call last):
[    9s]   File "/usr/bin/protoc-gen-nanopb", line 7, in <module>
[    9s]     from nanopb_generator import *
[    9s]   File "/usr/bin/nanopb_generator.py", line 53, in <module>
[    9s]     import proto
[    9s]   File "/usr/lib/python3.10/site-packages/proto/__init__.py", line 10, in <module>
[    9s]     import pkg_resources
[    9s] ModuleNotFoundError: No module named 'pkg_resources'
```

paolostivanin, 2023-01-30:
The issue with the fix from #841, is that there's now a conflict with https://github.com/googleapis/proto-plus-python , because both package will install `*/site-packages/proto/__init__.py`

PetteriAimonen, 2023-01-30:
Ok, I agree that the current CMake install of `proto` is pretty dirty. Some discussion is in #848 also.

I think the CMake script should be modified to install in a similar way as `pip install nanopb` does.

topazus, 2023-06-23:
> The issue with the fix from #841, is that there's now a conflict with https://github.com/googleapis/proto-plus-python , because both package will install `*/site-packages/proto/__init__.py`

@PetteriAimonen Hello! Recently, I try to make the packaging of nanopb into Fedora repository, the reviewer pointed out the nanopb package has the issue of the same install directory with the `python-proto-plus` package. see: https://bugzilla.redhat.com/show_bug.cgi?id=2216595#c4
```
[!]: Package does not own files or directories owned by other packages.
     Note: Dirs in package are owned also by: /usr/lib/python3.11/site-
     packages/proto(python3-proto-plus), /usr/lib/python3.11/site-
     packages/proto/__pycache__(python3-proto-plus)
```
Any suggestion or progress with this?

PetteriAimonen, 2023-06-23:
Suggestion would be to fix the CMake rules so that they install properly in a package called "nanopb" instead of "proto". E.g. imitate what `pip install nanopb` does currently.

The CMake rules have been contributed by users over the years and there is not really anyone actively maintaining them. I don't use CMake myself.

topazus, 2023-06-23:
@PetteriAimonen Maybe we can add the setup.py and pyproject.toml files (which exist in the uploaded source tarball of nanopb pypi project) into the proper location of the git repository, and use them for the `python-nanopb` package installation?

PetteriAimonen, 2023-06-23:
Maybe. Currently they somehow come from `extra/poetry/poetry_build.sh`, but I don't know the details on how that works either.

topazus, 2023-06-24:
@PetteriAimonen I will use the source tarball of nanopb pypi project to make python-nanopb package. I am not clear with python-nanopb package that includes files in `/usr/lib/python3.11/site-packages/nanopb/` directory. What is the relation of python-nanopb package with nanopb library package? So I can write the summary and description of the python-nanopb RPM package correctly.

PetteriAimonen, 2023-06-24:
The python part includes only the protoc-based generator, which converts .proto files to `.pb.h` files for inclusion in a project. It doesn't include the actual C source code for nanopb library.

The python part will depend on having either protoc (package "protobuf-compiler" or similar) or pip package grpcio-tools installed, in addition to python-protobuf library.

topazus, 2023-06-25:
@PetteriAimonen Thanks for the explanation of nanopb. I created the nanopb packages (nanopb, nanopb-devel, nanopb-python3) on Fedora repository, and will be available in a few days later.

Ref: https://src.fedoraproject.org/rpms/nanopb
https://packages.fedoraproject.org/pkgs/nanopb/

PetteriAimonen, 2023-11-11:
I think this should be fixed in 0.4.8.

closed 2023-11-11 by PetteriAimonen