nanopb/nanopb #849: error with pb.h

alirezainallo, 2023-01-07:
I want to build example/simple project for test but I give some error from <code>pb.h</code>:
[image]

PetteriAimonen, 2023-01-07:
Have you followed the recommendations above the comment, especially the part about enabling support for C11 standard in your compiler?

If this is MSVC, probably duplicate of #844.

alirezainallo, 2023-01-07:
> Have you followed the recommendations above the comment, especially the part about enabling support for C11 standard in your compiler?
> 
> If this is MSVC, probably duplicate of #844.

thank you [PetteriAimonen](https://github.com/PetteriAimonen). I solve it. my problem was for a function that have another name in my IDE: 
[image]
and compiler can't tell me about it because, it was a micro defined.

closed 2023-01-07 by alirezainallo