nanopb/nanopb #850: Try to build package using conan, got many errors

hpe-ykoehler, 2023-01-18:
When using conan, it seems the conanfile is out-of-date.  Still using cmake generators and the like.

Once I fixed those, I hit the issue where protoc was call, and then invoke_protoc try to call protoc again, but since it is not in the path, it fails.

I am unclear on how to solve that.  

I have tested that this was the issue by altering:

```
    status = invoke_protoc([sys.argv[0]] + args)
```

So that it can find itself, and that works, but I do not know of the consequence of such change.

I am unclear how/why protoc assume it is in the path, would need  help to solve the issue.

Here is the original error:

```
[ 20%] Generating nanopb_pb2.py
[100%] Built target protobuf-nanopb-static
Traceback (most recent call last):
  File "/work/conan-center-index/recipes/nanopb/all/build/Release/nanopb/../../../generator/protoc", line 44, in <module>
    status = invoke_protoc(['protoc'] + args)
  File "/work/conan-center-index/recipes/nanopb/all/generator/proto/_utils.py", line 44, in invoke_protoc
    return subprocess.call(argv)
  File "/usr/lib/python3.10/subprocess.py", line 345, in call
    with Popen(*popenargs, **kwargs) as p:
  File "/usr/lib/python3.10/subprocess.py", line 969, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.10/subprocess.py", line 1845, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'protoc'
gmake[2]: *** [nanopb/CMakeFiles/generate_nanopb_pb2.py.dir/build.make:73: nanopb/nanopb_pb2.py] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:100: nanopb/CMakeFiles/generate_nanopb_pb2.py.dir/all] Error 2
gmake: *** [Makefile:136: all] Error 2
```

PetteriAimonen, 2023-01-18:
Do you have Google's `protoc` compiler installed on your system, either through system package manager or with `pip install grpcio-tools`?

hpe-ykoehler, 2023-01-18:
No, I am building in a docker environment with minimal requirements for the build.  Does nanopb require protoc to build? what version then for 0.4.7? and which impl. protobuf official from google or a previous nanopb ?

If this is required we should state it as tools_required statement within conanfile.

PetteriAimonen, 2023-01-18:
Pretty much any version of Google's protoc should work, as long as it matches the `python-protobuf` version you have.
The `pip install --upgrade protobuf grpcio-tools` from README is what I recommend nowadays, though installing similar from distribution package manager is fine too.

There is no nanopb protoc, the `generator/protoc` is just a wrapper script to automatically give the correct paths to `protoc` as arguments.

The conan support has been contributed, I don't know much about conan myself.

hpe-ykoehler, 2023-01-18:
Is the protoc only avail from a python module?

What is the dependency on grpcio-tools?

I am not expert in neither conan or protobuf :-)  We are trying to adjust our build system towards conan and we have a requirement on nanopb, so I am trying to get that working.

I will submit PR to nanopb once I get a conanfile.py working with 1.56 and 2.0.0b8.

Also, I see you do not publish to conancenter, that would require some form of github action to trigger the conan build and upload to the conancenter, is that something you would be interested in? (No expert on github action either but could try to put it in place, makes me learn new things ;-))

PetteriAimonen, 2023-01-18:
protoc is available from many places, grpcio-tools Python package is one of them. You can also download a binary from Google, or install from system package manager.

PetteriAimonen, 2023-11-11:
I'm closing this issue because I'm not sure if there is anything here that needs to be changed. Feel free to reopen or add another issue with a more actionable description - I don't know anything about Conan myself.

closed 2023-11-11 by PetteriAimonen