nanopb/nanopb #856: Github Action to upload package to Conan Center

timmay54, 2023-02-28:
Being able to consume this package directly from conan.io would make it simpler for users to utilize this package. Since Conan is already baked into this repository, it seems like there could be a simple GitHub action created that would build, package, and upload several different configurations of this package to Conan center. 

I would be willing to assist with the creation of the githb action.

https://conan.io/center/search/nanopb

PetteriAimonen, 2023-02-28:
Sounds good.

If possible, make the workflow also test installation from conan.io, the pypi workflow could serve as a base:
https://github.com/nanopb/nanopb/blob/master/.github/workflows/pypi_publish.yml

Is there some conan.io account or project name that needs to be registered for nanopb?

luizfeldmann, 2023-10-29:
Any updates on this ?

How would you upload the builds to conan-center ? They host the recipes on their own repository and perform their own builds.

timmay54, 2024-01-23:
> Any updates on this ?
> 
> How would you upload the builds to conan-center ? They host the recipes on their own repository and perform their own builds.

https://github.com/timmay54/conan-center-index/tree/master

Ive began taking initial crack at getting this setup; would love an extra set of experienced eyes as this would be my first time attempting to contribute to conancenter's recipe list

as for the uploading piece; I am referring to a way to automate the management of this repo once added - a workflow that would upload/create PR's to the upstream conencenter repo for version bumps or things like that.