nanopb/nanopb #857: got "OSError: [WinError 145] 目录不是空的" when i use nanopb_generator.exe

albertyue, 2023-03-06:
Some exctption was raised when i use nanopb_generator.exe
here is output message:
```
 nanopb_generator.exe .\msg.proto
Traceback (most recent call last):
  File "nanopb_generator.py", line 2384, in <module>
  File "nanopb_generator.py", line 2251, in main_cli
  File "tempfile.py", line 807, in __exit__
  File "tempfile.py", line 811, in cleanup
  File "shutil.py", line 516, in rmtree
  File "shutil.py", line 404, in _rmtree_unsafe
  File "shutil.py", line 402, in _rmtree_unsafe
OSError: [WinError 145] 目录不是空的。: 'C:\\Users\\User\\AppData\\Local\\Temp\\tmpcdmoo4os'
[14868] Failed to execute script 'nanopb_generator' due to unhandled exception! 
```

PetteriAimonen, 2023-03-06:
Seems like some other process is creating a file in or locking the temporary directory created by the generator.

When you look at the directory `C:\\Users\\User\\AppData\\Local\\Temp\\tmpcdmoo4os` (name will be random every time), is there any file there?

PetteriAimonen, 2023-03-07:
Ok. I'll close this for now, but if you or someone else hits the same problem again, feel free to reopen.

closed 2023-03-07 by PetteriAimonen