nanopb/nanopb #859: first field in an optional submessage don't get encoded

Avaron165, 2023-03-15:
Whenever a submessage is set to optional the first field in the submessage is not encoded. This does apply whenever a field has the has_???? property.

So in proto2:

```
syntax = "proto2";

package update_protocol;

message SubMessage {
    required int32 test = 4;
    required int32 var3 = 6;
    required int32 var4 = 7;
}

message Test {
    required int32 var1 = 1;
    required int32 var2 = 2;
    optional SubMessage sub = 3;
}
```

the field SubMessage.test is not encoded,

or in proto3:

```
syntax = "proto3";

package update_protocol;

message SubMessage {
    int32 test = 4;
    int32 var3 = 6;
    int32 var4 = 7;
}

message Test {
    int32 var1 = 1;
    int32 var2 = 2;
    SubMessage sub = 3;
}
```

the Test message gets an implicit has_sub property and the field SubMessage.test is also not encoded....

Code Snippet where used:

```
   update_protocol_Test test = update_protocol_Test_init_zero;
    pb_ostream_t stream = pb_ostream_from_buffer(prototestbuffer, sizeof(prototestbuffer));

    test.var1 = 1;
    test.var2 = 2;    
    test.has_sub = true;
    test.sub.var3 = 3;
    test.sub.var4 = 4;
    test.sub.test = 5;

    OSAL_LOG_ERR("var1: %d",test.var1);
    OSAL_LOG_ERR("var2: %d",test.var2);
    OSAL_LOG_ERR("sub.var3: %d",test.sub.var3);
    OSAL_LOG_ERR("sub.var4: %d",test.sub.var4);
    OSAL_LOG_ERR("sub.test: %d",test.sub.test);

    bool status = pb_encode(&stream, update_protocol_Test_fields, &test);
    OSAL_LOG_ERR("Status: %d, Encode Length = %d",status, stream.bytes_written);

    OSAL_LOG_HEXDUMP_ERR(prototestbuffer,32,"buffer");

    update_protocol_Test test_decode = update_protocol_Test_init_zero;
    pb_istream_t istream = pb_istream_from_buffer(prototestbuffer, stream.bytes_written);
    status = pb_decode(&istream, update_protocol_Test_fields, &test_decode );

    OSAL_LOG_ERR("Status :%d",status);
    OSAL_LOG_ERR("var1: %d",test_decode.var1);
    OSAL_LOG_ERR("var2: %d",test_decode.var2);
    OSAL_LOG_ERR("sub.var3: %d",test_decode.sub.var3);
    OSAL_LOG_ERR("sub.var4: %d",test_decode.sub.var4);
    OSAL_LOG_ERR("sub.test: %d",test_decode.sub.test);

```

and generated output:

```
[00:07:43.972,000] <err> flash: var1: 1
[00:07:43.973,000] <err> flash: var2: 2
[00:07:43.974,000] <err> flash: sub.var3: 3
[00:07:43.974,000] <err> flash: sub.var4: 4
[00:07:43.975,000] <err> flash: sub.test: 5
[00:07:43.976,000] <err> flash: Status: 1, Encode Length = 16
[00:07:43.977,000] <err> flash: buffer
                                08 01 10 02 1a 0a 20 80  06 30 80 08 38 80 8a 03 |...... . .0..8...
                                00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
[00:07:43.980,000] <err> flash: Status :1
[00:07:43.981,000] <err> flash: var1: 1
[00:07:43.982,000] <err> flash: var2: 2
[00:07:43.982,000] <err> flash: sub.var3: 3
[00:07:43.983,000] <err> flash: sub.var4: 4
[00:07:43.984,000] <err> flash: sub.test: 0
```

PetteriAimonen, 2023-03-16:
Decoding the hex dump using https://protobuf-decoder.netlify.app/ shows that the fields get encoded, but with unexpected values.

Typically this means that the compiled version of `.pb.c` and the header `.pb.h` that is included do not match in some way. Make sure you have recompiled everything and there are no old file versions that could get accidentally included.

Another possibility is if you are using both C and C++ in the project, check if `sizeof(bool)` is different between C and C++ files. That would cause the structs to appear differently in memory when accessed by C code vs. C++.

Avaron165, 2023-03-16:
Hi, tanks a lot for your fast answer.

I tried recompiling all files and the problem still exists.
The value gets encoded as the same unexpected value, no matter what i try to set as value.

There are no c or c++ files mixed, all is plain C.

I had to include the gcc library as the 64bit shift opertators where not nativly available on my target.

Can you point me to some code where to debug this problem?

Avaron165, 2023-03-16:
Output:

```
[00:00:05.127,000] <err> flash: var1: 1
[00:00:05.127,000] <err> flash: var2: 2
[00:00:05.128,000] <err> flash: sub.var3: 3
[00:00:05.129,000] <err> flash: sub.var4: 4
[00:00:05.129,000] <err> flash: sub.tst: 5
[00:00:05.130,000] <err> flash: Status: 1, Encode Length = 16
[00:00:05.131,000] <err> flash: buffer
                                08 01 10 02 1a 0a 20 80  06 30 80 08 38 80 8c 03 |...... . .0..8...
                                00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00 |........ ........
[00:00:05.134,000] <err> flash: message struct
                                01 00 00 00 02 00 00 00  01 05 00 00 00 03 00 00 |........ ........
                                00 04 00 00 00                                   |.....
[00:00:05.138,000] <err> flash: message descriptor
                                04 11 00 40 08 11 04 40  0d 18 01 40 0c 00 0c 00 |...@...@ ...@....
                                00 00 00 00 c0 b9 00 00  ec b9 00 00 00 00 00 00 |........ ........
                                00 00 00 00 03 00 00 00  03 00 00 00 a4 b9 00 00 |........ ........
                                00 00 00 00 30 2e 31 2e  30 2d 30 00 45 55 46 45 |....0.1. 0-0.EUFE
[00:00:05.144,000] <err> flash: Status :1
[00:00:05.145,000] <err> flash: var1: 1
[00:00:05.145,000] <err> flash: var2: 2
[00:00:05.146,000] <err> flash: sub.var3: 3
[00:00:05.147,000] <err> flash: sub.var4: 4
[00:00:05.147,000] <err> flash: sub.tst: 0
```

PetteriAimonen, 2023-03-16:
Ok, so let's see.

Message struct:

```
01 00 00 00 // var1
02 00 00 00 // var2
01          // has_sub
05 00 00 00 // test
03 00 00 00 // var3
04 00 00 00 // var4
```

Descriptor (comparing to [pb.h definition](https://github.com/nanopb/nanopb/blob/master/pb.h#L728-L754)):

```
04 11 00 40  Field 1, proto3 varint, data_offset 0, data_size 4

08 11 04 40  Field 2, proto3 varint, data_offset 4, data_size 4

0d 18 01 40  Field 3, optional submessage, size_offset 4
0c 00 0c 00  data_offset 12, data_size 12

00 00 00 00  End of descriptor

```

So yeah, the descriptor says the submessage begins at offset 12, so there is 4 bytes received for the `bool`. But the message struct only has 1 byte reserved. This means that there is some significant difference between how the `.pb.c` file is compiled and how your main `.c` file is compiled.

Typically this would be because of something wrong with compiler configuration, e.g. if you have different compilation options for different files and they make the structs incompatible. Something like `#pragma pack` can also affect it.

Avaron165, 2023-03-16:
Thank you so much!

It turned out that it was a #pragma pack(1) directive in some other header file included.

Changes this to #pragma pack(push,1) and #pragma pack(pop) at the end and now eveything is working as expected.

closed 2023-03-16 by Avaron165