nanopb/nanopb #861: Encoding strings with strcpy(), but type is pb_callback_t

LiamvanLeusden, 2023-03-23:
Hello everyone and I'm sorry if this is a dumb question. I want to use NanoPB using an STM32 and PlatformIO. I added the NanoPB module to my project and compiling proto files works well.

Here is my .proto file:

```
syntax = "proto3";

message TestDataStruct {
    int32 testint = 1;
    double testdouble = 2;
    float testfloat = 3;
    string teststring = 4;
}
```

If I go to set the values in a message to be sent, all variable types work, except the string. It is not a character array but a `pb_callback_t`. In the example found in [nanopb/examples/platformio/src/pio_with_options.c](https://github.com/nanopb/nanopb/blob/master/examples/platformio/src/pio_with_options.c) they just copy a string into the field with strcpy. This would be pefect for my use case, and would look like this:

```
messageToBeSent.testint = 12;
messageToBeSent.testdouble = 23.31;
messageToBeSent.testfloat = 60.12435;
strcpy(messageToBeSent.teststring, "Hello World");
```

A `pb_callback_t` contains a void pointer argument called `arg`.  I tried doing this `messageToBeSent.teststring.arg = (void*)"Hello World!";`, and reading it with `(char*)messageToBeSent.teststring.arg` seems to work, but if encoded like this:

```
uint8_t buffer[50];
pb_ostream_t stream = pb_ostream_from_buffer(buffer, 50);
pb_encode(&stream, TestDataStruct_fields, &messageToBeSent);
```
And decoded like this;

```
TestDataStruct result = TestDataStruct_init_zero;
uint8_t buffer1[50];
pb_istream_t stream1 = pb_istream_from_buffer(buffer1, 50);
pb_decode(&stream1, TestDataStruct_fields, &result);
```
The result data struct (`result.teststring.arg`) does not include the contents of the string and not even the memory address of the old string. The other data types survive the encoding and decoding and can be read from the result data struct.

LiamvanLeusden, 2023-03-23:
Thank you! I can personally close this issue because a maximum size is suitable for my application. However, some people may encounter this issue with a need for dynamic strings. Is there an easy way to encode them? Do you have to use `message.func.encode()`?

PetteriAimonen, 2023-03-23:
The options when maximum size is not known are either dynamic allocation with `PB_ENABLE_MALLOC` or specifying a callback function.