nanopb/nanopb #863: In Compatibility between C generated message and Python decoder

dmliquidity, 2023-04-09:
I have a very simple use case which I cannot get to work.
TLDR;
The c code using nanopb generates b'\x04Cam1\x070.0.0.0' encoding the same message in python generates b'\n\x04Cam1\x12\x070.0.0.0' the python server thus cannot decode the C protobuff stream.

Proto file.
`syntax = "proto3";
package espcam32;

message RegisterRequest {
  string name = 1;
  string ip = 2;
}

message ResponseCode {
  int32 responseCode = 1;
}

service RegisterService {
  rpc Register (RegisterRequest) returns (ResponseCode);
}
`
Message is encoded and sent on the hardware.
`		const char* name = "Cam1";
		const char* ip = "0.0.0.0";
		uint8_t buffer[256];			

		espcam32_RegisterRequest message = espcam32_RegisterRequest_init_zero;
		message.name.arg = (void *)name;
		message.name.funcs.encode = &encode_string;
		message.ip.arg = (void *)ip;
		message.ip.funcs.encode = &encode_string;
		Serial.println("Encodeing");
		// THIS CODE BREAKS HERE> Core  0 panic'ed (LoadProhibited). Exception was unhandled.
		pb_ostream_t stream = pb_ostream_from_buffer(buffer, sizeof(buffer));
		bool status = pb_encode(&stream, espcam32_RegisterRequest_fields, &message);
		if (!status)
		{
			Serial.println("Failed to encode");
		} else {
			Serial.println("Encoded");
			for(int i = 0; i<stream.bytes_written; i++){
 				Serial.printf("%02X",buffer[i]);
			}
		}

		if (!client.connected()) {
			if (!client.connect(serverIp, PORT_NUMBER)) {
				Serial.println("Failed to connect to server");
			}
        }

		if (client.connected()) {
			client.write((const uint8_t*)buffer, stream.bytes_written);
			client.flush();
			Serial.println("Data sent to server");
		}`
Message is Decoded on the python server.

`def handle_client(conn):
    # Receive the message from the client
    data = conn.recv(1024)
    if not data:
        return
    decoded_data = data.decode('utf-8')
    print(f"Data from client: {data}")
    print(f"Data from client: {decoded_data}")
    # Parse the message using protobuf
	# OK THIS IS NOT DECODEING
	# data = b'\x04Cam1\x070.0.0.0'
	# So the fields are there 0x04 and 0x07 as well as the strings Cam1 and 0.0.0.0 that were sent.
	# However its Missing the \x12 expected by python.
	# Enviroment Variables set to declare -x PROTOCOL_BUFFERS_PYTHON_IMPLEMENTATION="cpp"
	# See https://protobuf.dev/reference/python/python-generated/#sharing-messages
        #No idea if that does anything....
    request = esp32_cam_pb2.RegisterRequest()
    request.ParseFromString(data) #<= Run it in the console => <string>:1: RuntimeWarning: Unexpected end-group tag: Not all data was converted
    print(f"Received message from client: {request.name}, {request.ip}")
    # Send the response back to the client
    response = esp32_cam_pb2.ResponseCode(responseCode=200)
    conn.send(response.SerializeToString())
    conn.close()`

If you encode and decode the request in Python you can see what it's expecting.
`
        request = esp32_cam_pb2.RegisterRequest()
	request.name = "Cam1"
	request.ip = "0.0.0.0"
    # This thing encodes and decodes as 
	# b'\n\x04Cam1\x12\x070.0.0.0'
	# Clearly the Python and C versions are not compatible. 
	# The C version is sending b'\x04Cam1\x070.0.0.0'
        # it's missing the \x12
	print(request)
	print(request.SerializeToString())`
	
It late here and I have been banging my head against this most of the day, sorry for the code dump.
I'll get the whole  code into a public repo and link here.

dmliquidity, 2023-04-09:
Full example here.
https://github.com/dmliquidity/ai-esp32cam-python-grpc

PetteriAimonen, 2023-04-09:
> decoded_data = data.decode('utf-8')

Protobuf data is binary format, you should not do any utf-8 decoding or string formatting on it.

Is the hex dump that you are printing correct, i.e. does it parse with e.g. https://protobuf-decoder.netlify.app/

dmliquidity, 2023-04-10:
Yeap, I was getting desperate there with the utf-8 decoding.

```
	# Receive the message from the client
	data = conn.recv(1024)
	if not data:
		return
	print(f"Data from client: {data}")
	# Data from client: b'\x04Cam1\x070.0.0.0'
	print(f"Data from client as Hex: {data.hex()}")
	# Data from client as Hex: 0443616d3107302e302e302e30
	# ^ Does not decode at https://protobuf-decoder.netlify.app/
```
The Python Version. Decodes correctly. 
```
	request = esp32_cam_pb2.RegisterRequest()
	request.name = "Cam1"
	request.ip = "0.0.0.0"
	# This thing encodes and decodes as
	print(request)
	print(f"Python Encoded as String {request.SerializeToString()}")
	# Python Encoded as String b'\n\x04Cam1\x12\x070.0.0.0'
	print(f"Python Encoded as Hex {request.SerializeToString().hex()}")
	# Python Encoded as Hex 0a0443616d311207302e302e302e30
	# ^ Decodes @ https://protobuf-decoder.netlify.app/ Correctly
```
I'll take another look as the ESPCam code and get a hex dump prior to transmission and check that.

dmliquidity, 2023-04-10:
Found the issue the variable length encoder was incorrect.

Incorrect Code.
```
bool encode_string(pb_ostream_t *stream, const pb_field_t *field, void * const *arg) {
   const char* str = (const char*)(*arg);
   return pb_encode_string(stream, (uint8_t *)str, strlen(str));
}
```
Correct code.
```
bool encode_string(pb_ostream_t *stream, const pb_field_t *field, void * const *arg) {
	const char* str = (const char*)(*arg);
	if (!pb_encode_tag_for_field(stream, field))
		return false;
    return pb_encode_string(stream, (uint8_t *)str, strlen(str));
}
```

closed 2023-04-10 by dmliquidity