nanopb/nanopb #864: Optimizing for speed

JoakimKjellberg, 2023-04-12:
Hi. I have to say that like this library and appreciate your efforts maintain it!

I'm running an application where encoding speed is important, but number of bytes sent on Ethernet is not. I saw your response to this question [here](https://stackoverflow.com/questions/73146576/nanopb-how-to-optimize-encoding-for-runtime-speed). Since I'm using a lot of sub-messages, the encoding is relatively slow.

Do you think it would be possible to create a mode where the encoded data has a pre-defined size? Instead of using varints and such, one could just pack the entire uint32 (and other types), even if the value is small. Instead if checking if a sub-message contains non-zero values, one could just send all values, even if they are zero. Perhaps such an encode function could then be generated and stored in the generated `<proto>.pb.h` file.

**What should the feature do?**
Optimize for encoding speed by knowing the size of the data beforehand.

**In what situation would the feature be useful?**
When encoding speed is important, but data size is not.

PetteriAimonen, 2023-04-12:
I agree that nanopb is currently rather slow.

I have been considering generating message-specific encoder and decoder functions that would avoid a lot of the overhead of iterating through the descriptors. A memcpy approach similar to #550 could speed up submessage encoding. But I probably won't have time for it in any near future.

As for optimizations that change the wire format (with expanded varints or submessage padding), I think they are too specific to be included in the main library.

For highest speed, protobuf format is simple enough to encode even fully manually. You could consider doing that in a callback for some of the submessages that are more numerous.

JoakimKjellberg, 2023-04-12:
Thanks for the quick response! I will have a look at #550 when I have time.

Do you have ideas on how I can use functions from nanopb to manually encode expanded types and submessages? Because when I read the description for [pb_encode_submessage](https://jpa.kapsi.fi/nanopb/docs/reference.html#pb_encode_submessage) I see that it encodes the message twice. But since I want to use a static size, that shouldn't be necessary. If you have the time to point me in the right direction, that would be highly appreciated!

PetteriAimonen, 2023-04-12:
If you know the size already, you can call `pb_encode_varint` to write it and then `pb_encode` to encode the message data, or encode the message data manually to avoid the descriptor parsing overhead.

Compare with [pb_encode_submessage](https://github.com/nanopb/nanopb/blob/master/pb_encode.c#L723-L771) source where up to line 738 is to determine the size and the latter part is to encode for real.

closed 2023-11-11 by PetteriAimonen