nanopb/nanopb #866: Generator not picking up options file in same folder

infn-ke, 2023-04-19:
I'm running cmake `nanopb_generate_cpp`  to generate my pb files. This works fine, but the options file is not picked up. The options file resides in the same folder as the proto definition. I enabled verbose output and see the following

Options file not found: proto/example/example.options
Options for proto/example/example.proto: proto3: true

Is it dropped due to not found or due to incompatible with the proto file? I'm using a `package` name in the proto but not completely sure how to express that in the options file.

PetteriAimonen, 2023-04-19:
Sounds like whole file not found.

It is probably something with the CMake rules, IIRC some of them copy the files over to different path or something.

Does same happen if you call `generator/nanopb_generator.py proto/example/example.proto` directly?

infn-ke, 2023-04-19:
> Does same happen if you call `generator/nanopb_generator.py proto/example/example.proto` directly?

Invoking nanopb_generator.py directly works, then it correctly picks up the options file. What did I do wrong? Do I need to pass some option  to ` nanopb_generate_cpp`? Or I hit a bug in the cmake recipes?

PetteriAimonen, 2023-04-19:
I don't know, the CMake rules have all been contributed by users - my own CMake experience is very limited.

Adding `simple.options` to `examples/cmake_simple` appears to work correctly though. Can you make a minimal example that fails?

infn-ke, 2023-04-19:
Know I see, when I set `NANOPB_GENERATE_CPP_APPEND_PATH` to `FALSE` it causes the options file to not be found. Not possible to combine options and this parameter?

infn-ke, 2023-04-19:
> Know I see, when I set `NANOPB_GENERATE_CPP_APPEND_PATH` to `FALSE` it causes the options file to not be found. Not possible to combine options and this parameter? I would like the generated files to end up in proto build folder instead of just the current build dir.

Figured out one way to workaround it; 

```cmake
set(NANOPB_OPTIONS "-v --options-path ${PROTO_SOURCE_DIR}")
```

Not sure this is the correct way to fix this. Please advice.

PetteriAimonen, 2023-04-20:
Sounds reasonable.

Generator not finding options files has been a recurring problem, as protoc doesn't inform plugins of the actual path of files.

closed 2023-04-20 by infn-ke