nanopb/nanopb #870: Nanopb example simple doesn't work on Raspberry with "make" generator-bin/protoc: 2: Syntax error: "(" unexpected

GaetanGodart, 2023-04-27:
Hello (Petteri),

I tried running the simple example on a Raspberry 3b+ but it failled to compile.

I solved it (maybe there is a simpler fix), I am posting here to try to help someone having the same issue. I hope this is the right place. I also think it could be nice to post it on the google group. Tell me if that is the right way of doing it, I am also thinking of posting an issue with fix regarding the network example and posting another example that I made.

**Environment :**
Raspberry 3 model B+
OS Raspbian 10.2.1-6+rpi1
Gcc 10.2.1
Cmake 3.18.4
nanopb nanopb-0.4.7-linux-x86

**Steps to reproduce the issue :**
1. I downloaded nanopb for linux on my Raspberry 3b+.
2. I tried to run the simple example by following the instruction on the read me (just going into the corresponding file and typing "make").
3. I added the "generator" folder to the path.
4. I tried the "make" command once more.
5. I tried the command "make all".

**What happens :**
It's simpler to just show my terminal:

```
usr@raspberrypi:~/Documents/nanopb-0.4.7-linux-x86/examples/simple $ ls
Makefile  README.txt  simple.c  simple.proto

usr@raspberrypi:~/Documents/nanopb-0.4.7-linux-x86/examples/simple $ make
../../generator-bin/protoc  --nanopb_out=. simple.proto
../../generator-bin/protoc: 2: Syntax error: "(" unexpected
make: *** [Makefile:21: simple.pb.c] Error 2

usr@raspberrypi:~/Documents/nanopb-0.4.7-linux-x86/examples/simple $ export PATH="/home/usr/Documents/nanopb-0.4.7-linux-x86/generator:$PATH"

usr@raspberrypi:~/Documents/nanopb-0.4.7-linux-x86/examples/simple $ make
../../generator-bin/protoc  --nanopb_out=. simple.proto
../../generator-bin/protoc: 2: Syntax error: "(" unexpected
make: *** [Makefile:21: simple.pb.c] Error 2

usr@raspberrypi:~/Documents/nanopb-0.4.7-linux-x86/examples/simple $ ls
Makefile  README.txt  simple.c  simple.proto

@raspberrypi:~/Documents/nanopb-0.4.7-linux-x86/examples/simple $ make all
make: *** No rule to make target 'all'.  Stop.
```

**What should happen :**
SImple.c should be compiled.

**The fix I found :**
I changed the make file to:

```
# Include the nanopb provided Makefile rules
include ../../extra/nanopb.mk
PROTOCPATH := PutYourPathToTheGeneratorFileHere

# Compiler flags to enable all warnings & debug info
CFLAGS = -Wall -Werror -g -O0
CFLAGS += "-I$(NANOPB_DIR)"

# C source code files that are required
CSRC  = simple.c                   # The main program
CSRC += simple.pb.c                # The compiled protocol definition
CSRC += $(NANOPB_DIR)/pb_encode.c  # The nanopb encoder
CSRC += $(NANOPB_DIR)/pb_decode.c  # The nanopb decoder
CSRC += $(NANOPB_DIR)/pb_common.c  # The nanopb common parts

# Build rule for the main program
simple: $(CSRC)
	$(CC) $(CFLAGS) -osimple $(CSRC)

# Build rule for the protocol
proto:
	$(PROTOCPATH)/protoc --nanopb_out=. simple.proto
#simple.pb.c: simple.proto
#	$(PROTOC) $(PROTOC_OPTS) --nanopb_out=. simple.proto

all: proto simple

clear:
	rm -f simple simple.pb.c simple.pb.h
```

Here is what I got in the terminal:

```
usr@raspberrypi:~/Documents/nanopb-0.4.7-linux-x86/examples/simpleFix $ ls
Makefile  README.txt  simple.c  simple.proto

usr@raspberrypi:~/Documents/nanopb-0.4.7-linux-x86/examples/simpleFix $ make all
/home/usr/Documents/nanopb-0.4.7-linux-x86/generator/protoc --nanopb_out=. simple.proto
cc -Wall -Werror -g -O0 "-I../.." -osimple simple.c                    simple.pb.c                 ../../pb_encode.c   ../../pb_decode.c   ../../pb_common.c  

usr@raspberrypi:~/Documents/nanopb-0.4.7-linux-x86/examples/simpleFix $ ls
Makefile  README.txt  simple  simple.c  simple.pb.c  simple.pb.h  simple.proto

usr@raspberrypi:~/Documents/nanopb-0.4.7-linux-x86/examples/simpleFix $ ./simple
Your lucky number was 13!
```

I hope the formating is good and that this can help someone having the same issue.

PetteriAimonen, 2023-04-27:
The x86 binary package will only work on platforms with x86 processor. Raspberry Pi has ARM processor architecture.

Try the source package, it should work the same except you may need to install dependencies from apt-get:

    apt-get install python3-protobuf protobuf-compiler

GaetanGodart, 2023-04-27:
Thanks for your answer.
It does fix the issue.

I did not have to install dependencies, I did already do pip install grpcio and grpcio_tools when trying grpc with python so that may be why.

PetteriAimonen, 2023-04-27:
Yeah, that brings the newest versions from pip instead.

closed 2023-04-27 by PetteriAimonen