nanopb/nanopb #872: Is there a way to encode message to a buffer WITHOUT stream?

424778940z, 2023-05-04:
In our use case we have to add headers to each data packet we send through, it make more sense to only use nanopb to encode message to a buffer, then we handle sending ourselfs.
I checked documents, seems all pb_encode_xxx requires a pb_ostream_t. Handle communications inside callbacks is really odd.

Did I missed something? Is there a way to only to just encode to a buffer? (and decode from a buffer to message?)

apcountryman, 2023-05-04:
[`pb_ostream_from_buffer()`](https://jpa.kapsi.fi/nanopb/docs/reference.html#pb_ostream_from_buffer) is used to create a `pb_ostream_t` that writes to the buffer. [`pb_istream_from_buffer()`](https://jpa.kapsi.fi/nanopb/docs/reference.html#pb_istream_from_buffer) is used to create a `pb_istream_t` that reads from a buffer.

424778940z, 2023-05-04:
thx for the quick reply, i will check them out

424778940z, 2023-05-04:
solved, closing

closed 2023-05-04 by 424778940z