nanopb/nanopb #873: missing call in custom_command for MSVC

philippe44, 2023-05-10:
I think the custom_command sof "FindNanopb.cmake" are missing a "call" prefix that causes these to never return properly in MSVC. As a result, .pb.c and .pb.h are not built. I suggest adding something like that
```
add_custom_command(
      OUTPUT "${CMAKE_CURRENT_BINARY_DIR}/${FIL_PATH_REL}/${FIL_WE}.pb.c"
             "${CMAKE_CURRENT_BINARY_DIR}/${FIL_PATH_REL}/${FIL_WE}.pb.h"
      COMMAND ${CALL_PREFIX} ${PROTOBUF_PROTOC_EXECUTABLE}
      ARGS -I${GENERATOR_PATH} -I${GENERATOR_CORE_DIR}
           -I${CMAKE_CURRENT_BINARY_DIR} ${_nanopb_include_path}
           --plugin=protoc-gen-nanopb=${NANOPB_GENERATOR_PLUGIN}
           ${NANOPB_OPT_STRING}
           ${ABS_FIL}
      DEPENDS ${ABS_FIL} ${GENERATOR_CORE_PYTHON_SRC}
           ${ABS_OPT_FIL} ${NANOPB_DEPENDS}
      COMMENT "Running C++ protocol buffer compiler using nanopb plugin on ${FIL}"
      VERBATIM )
```
Where CALL_PREFIX is defined to "call" under MSVC and nothing otherwise

PetteriAimonen, 2023-05-10:
Thanks for the report.

Unfortunately I don't have the time or experience to maintain the CMake rules either..

philippe44, 2023-05-10:
I can offer a PR, but it might not be the most elegant 😄

PetteriAimonen, 2023-05-10:
What this would really need is a github workflow that tests it.

PetteriAimonen, 2023-11-11:
Fix published in nanopb-0.4.8.

closed 2023-11-11 by PetteriAimonen