nanopb/nanopb #877: nanopb_pb2.py placed in the wrong folder upon pip3 install nanopb

ghpqans, 2023-06-20:
As mentioned in the Issue https://github.com/nanopb/nanopb/issues/629, the mentioned module is still placed in the wrong folder in version 0.47.
`.venv/lib/python3.10/site-packages/nanopb/generator/proto` instead of `.venv/lib/python3.10/site-packages/`.
The generated file with the nanopb message definitions contrains the import statement: import nanopb_pb2 as nanopb__pb2` which attempts to import the `nanopb_pb2` module from the top-level site-packages.
This does not match the location where the file is found.

PetteriAimonen, 2023-06-20:
I don't see a reason why the second path would be correct.

If I understand correctly, this is about the import `protoc` generates when using your `.proto` file with Python library?
In that case there are two options:

1. Use the separate `.options` file for nanopb options. This avoids any nanopb reference in the `.proto` file.
2. Generate `nanopb_pb2.py` from `nanopb.proto` alongside your own proto files.

ghpqans, 2023-06-20:
Sorry I misunderstood that. So the generated import statement is from the following import in the `.proto` file? I thought it was some line which is added per se.

```
syntax = "proto3";

import "nanopb.proto";

option (nanopb_fileopt).proto3_singular_msgs = true;
option (nanopb_fileopt).anonymous_oneof      = true;
option (nanopb_fileopt).fixed_length         = true;
option (nanopb_fileopt).long_names           = false;
```

ghpqans, 2023-06-20:
I think maybe the point was that I specified an include path in the `protoc` command line:
`-I=.venv/lib/python3.10/site-packages/nanopb/generator/proto`.
Therefore `protoc` generated the `nanopb_pb2.py` there and not in the output directory and not in the path specified by `--python_out=some_dir`.

ghpqans, 2023-06-20:
So, why is the output file generated in a directory I did not specify?

ghpqans, 2023-06-20:
Actually, the output for `nanopb.proto` is generated but in the source directory and not in the target directory.
The source directory was specified by the `-I=.venv/lib/python3.10/site-packages/nanopb/generator/proto` flag and there the generated file is found. 
However, this is not the target directory. The target is specified by `--python_out=some_dir` and only the generated file corresponding to the top-level `.proto` file will be found there.

PetteriAimonen, 2023-06-20:
> only the generated file corresponding to the top-level .proto file will be found there.

If you run `protoc ..... .venv/lib/python3.10/site-packages/nanopb/generator/proto/nanopb.proto` it will probably generate nanopb_pb2.py into your output directory.

PetteriAimonen, 2023-10-19:
Closing off as this sounds more like usage problem than a bug.

closed 2023-10-19 by PetteriAimonen