nanopb/nanopb #879: PB_FIELD_32BIT roundtrip unexpected behaviour

Roysten, 2023-06-30:
Hello,

first of all thank you for this project. I have been using it in some projects and it was nice to work with.

I am encountering some unexpected behaviour when using PB_FIELD_32BIT. I created a simple proto file and some C code to demonstrate the problem.
```protobuf
syntax = "proto3";

import "nanopb.proto";

enum FIELD
{
   FIELD_UNSPECIFIED = 0;
   FIELD_ID = 1;
   FIELD_NAME = 2;
   FIELD_GUID = 3;
}

message REQUEST_METADATA
{
   sfixed32 id = 1;
}

message GET_RESOURCE_INFO
{
   message PARAMS
   {
      repeated fixed32 ids = 1 [(nanopb).max_count = 1000];
      repeated FIELD fields = 2 [(nanopb).max_count = 4];
   }
   REQUEST_METADATA metadata = 1;
   PARAMS params = 2;
}
```
```C
#include <stdbool.h>
#include <inttypes.h>
#include <string.h>
#include <stdlib.h>
#include <stdio.h>

#include <pb_encode.h>
#include <pb_decode.h>

#include "test.pb.h"

int main()
{
   GET_RESOURCE_INFO request;
   memset(&request, 0, sizeof(request));

   request.metadata.id = 3;
   request.has_metadata = true;
   request.params.ids_count = 4;
   request.params.ids[0] = 1;
   request.params.ids[1] = 2;
   request.params.ids[2] = 3;
   request.params.ids[3] = 4;
   request.params.fields_count = 1;
   request.params.fields[0] = FIELD_FIELD_GUID;
   request.has_params = true;

   size_t BUF_LEN = 5 * 1024 * 1024;
   uint8_t *buf = malloc(BUF_LEN);
   memset(buf, 0, BUF_LEN);

   printf("Nanopb version: %s\n", NANOPB_VERSION);
   printf("PB header version: %d\n", PB_PROTO_HEADER_VERSION);

   pb_ostream_t ostream = pb_ostream_from_buffer(buf, BUF_LEN);
   bool encode_ok = pb_encode(&ostream, GET_RESOURCE_INFO_fields, &request);
   printf("Encode ok: %d\n", encode_ok);

   GET_RESOURCE_INFO decoded = {0};
   pb_istream_t istream = pb_istream_from_buffer(buf, ostream.bytes_written);
   bool decode_ok = pb_decode(&istream, GET_RESOURCE_INFO_fields, &decoded);
   printf("Decode ok: %d\n", decode_ok);

   printf("Has metadata: %d, has params: %d\n", decoded.has_metadata, decoded.has_params);
   return 0;
}
```

Output when NOT using PB_FIELD_32BIT:
```console
$ ./test
Nanopb version: nanopb-0.4.7
PB header version: 40
Encode ok: 1
Decode ok: 1
Has metadata: 1, has params: 1
```
Output with PB_FIELD_32BIT:
```console
Nanopb version: nanopb-0.4.7
PB header version: 40
Encode ok: 1
Decode ok: 1
Has metadata: 1, has params: 0
```
I expected both outputs to be the same. I set the PB_FIELD_32BIT option by passing -DPB_FIELD_32BIT to gcc. Am I doing something wrong here or is this a bug?

Thanks in advance,
Roy

PetteriAimonen, 2023-06-30:
Make sure you have recompiled all source files and that `PB_FIELD_32BIT` is defined on the compiler command line or in pb.h, so that it applies to all files.

Otherwise it can happen that the `test.pb.c` file as compiled thinks that the `pb_size_t` fields in struct are different size. Then it will end up writing data to wrong place in the struct.

PetteriAimonen, 2023-06-30:
For reference, I get same output both ways:

<pre>
petteri@raichu:~/tmp6$ gcc -Wall test.pb.c main.c ~/nanopb/pb_encode.c ~/nanopb/pb_decode.c ~/nanopb/pb_common.c -I ~/nanopb
petteri@raichu:~/tmp6$ ./a.out 
Nanopb version: nanopb-0.4.8-dev
PB header version: 40
Encode ok: 1
Decode ok: 1
Has metadata: 1, has params: 1
petteri@raichu:~/tmp6$ gcc -Wall -DPB_FIELD_32BIT test.pb.c main.c ~/nanopb/pb_encode.c ~/nanopb/pb_decode.c ~/nanopb/pb_common.c -I ~/nanopb
petteri@raichu:~/tmp6$ ./a.out 
Nanopb version: nanopb-0.4.8-dev
PB header version: 40
Encode ok: 1
Decode ok: 1
Has metadata: 1, has params: 1
</pre>

Roysten, 2023-07-01:
Thank you, indeed this was the problem. Sorry for wasting your time.

Is it maybe a good idea to add a function to one of the .c files with which one can request what field size the library was compiled with? This way users have a way to validate their settings.

PetteriAimonen, 2023-07-01:
I have considered something like that, but it is hard to make actually reliable indication.

Most practical would be to include `sizeof(msg)` in the descriptor in `myproto.pb.c` and pass that also to `pb_encode()`/`pb_decode()`. This still wouldn't detect difference of settings between `pb_encode.c`/`pb_decode.c` though, only between user code and the compiled proto file, so that would need another mechanism for validation.

Yet another way would be to name the symbols using some suffix generated from compilation settings using macros. But I feel that gets a bit too hacky.

closed 2023-07-01 by PetteriAimonen