nanopb/nanopb #881: Recursive message definitions are broken

mkruskal-google, 2023-07-10:
I work on [Protobuf](https://github.com/protocolbuffers/protobuf), and I'm currently trying to add a new recursive field definition to `descriptor.proto`.  This is being blocked by https://github.com/nanopb/nanopb/issues/493, which causes infinite recursion in nanopb.  We're currently using an older version of nanopb in Google, but AFAICT from that issue, these types of protos still aren't supported.  It seems like the resolution there was to provide a nicer error message, but not actually support recursive messages.

Is there a reason this isn't possible in nanopb?  Recursive messages are valid in Protobuf, so this is a major deviation between the two libraries

PetteriAimonen, 2023-07-10:
First question would be, do you actually need nanopb access to descriptor.proto structures? It's quite rare to process them in the kinds of tasks nanopb is used for.

But regarding recursive messages, they work if you define options `(nanopb).type` as either `FT_CALLBACK` or `FT_POINTER` (or `FT_IGNORE`, but that just skips the field). See tests/cyclic_messages.

mkruskal-google, 2023-07-10:
*nanopb* needs access to descriptor.proto IIUC.  You currently have a fork of it in your repo, but within google we use a shared version.  We also can't define custom options without taking on a dependency to nanopb, which would cause cyclic dependency issues.

PetteriAimonen, 2023-07-11:
Sure, descriptor.proto is used by the nanopb generator to parse the .proto files. However, this uses the python-protobuf library and happens on a regular computer.

The C library as used on embedded systems usually does not see the descriptor.proto structures at all. But if there was a specific purpose to parse them on the embedded side, that is possible, it just needs a callback or pointer at some point of the cycle. Static allocation cannot by its very nature represent cyclic structures.

PetteriAimonen, 2023-07-11:
Yeah, automatic detection of cycles & conversion to FT_CALLBACK sounds reasonable. Maybe even worth a backport to 0.3.9.x series if that is what you are on. It's just a question on which reference to convert to callback - would need some reasonable heuristic, and also need to check whether there are already options specified. I'm not very eager to work on this myself currently.

Normally nanopb_generator [excludes](https://github.com/nanopb/nanopb/blob/master/generator/nanopb_generator.py#L2435) `google/protobuf/descriptor.proto` from the generated includes, so that there is no need to generate a `.pb.h` for it. Possibly if you have descriptor.proto in a different relative path, it doesn't get matched? It can be added to excludes manually with the `-x` option.

For cases where someone would have actual need to access descriptor.proto structures using nanopb, they would want to use options anyway - using the defaults for every field is just not reasonable. For that case, the options would be placed in a separate file and `--options-file` used to apply them.

mkruskal-google, 2023-07-11:
hah yep you were correct, we weren't properly excluding our internal `descriptor.proto`.  Adding that does seem to fix the issue I'm seeing, so at least this is unblocked.  Leaving the bug open since the (lower priority) root issue is still there though

mkruskal-google, 2023-07-11:
Hah, using an options file is a solid workaround!  That unblocks us, but I think this is still a reasonable feature request for the future

PetteriAimonen, 2023-11-11:
Fix published in nanopb-0.4.8.

closed 2023-11-11 by PetteriAimonen