nanopb/nanopb #883: Need help with repeated field and callbacks

TSMotter, 2023-07-16:
Hello,
I have what I believe to be a fairly simple `protobuf` message specification but I'm having a hard time understanding how to deal with repeated fields callbacks.

When I try to decode the binary encoded message with `protoc --decode Batch protocol.proto < binary` I get `Failed to parse input.`

I hope some one can provide some help.

I'll provide the following information below:
**1. My ".proto"
2. My code snippet for encoding process
3. My callbacks
4. The output I observe due to the printf's**

____

**1. This is my proto:**
  ```proto
  syntax = "proto3";
  message Batch {
      message Sample {
          string name = 2;
          uint32 frequency = 3;
          double time = 4;
          double value = 5;
      }
      repeated Sample items = 1;
  }
  ```
**2. This is my initialization for encoding just for testing:**
  ```C
      RingBuffer_Init(&common_rb, common_buffer, sizeof(uint8_t), COMMON_BUFFER_SIZE);
      pb_ostream_t oStream = {&custom_ostream_callback, (void *) &common_rb, COMMON_BUFFER_SIZE, 0};
      Batch_Sample samples[2] = {{.name.arg          = "channel_2000_200_1",
                                  .name.funcs.encode = &custom_field_encoding_callback,
                                  .frequency         = 200,
                                  .time              = 0.051526546478271484,
                                  .value             = 0.31812476943300294},
                                 {.name.arg          = "channel_2000_200_1",
                                  .name.funcs.encode = &custom_field_encoding_callback,
                                  .frequency         = 200,
                                  .time              = 0.056981563568115234,
                                  .value             = 0.3504258283522246}};
      Batch batch_e = {.items.arg          = samples,
                       .items.funcs.encode = &custom_repeated_encoding_callback};
      pb_encode(&oStream, Batch_fields, &batch_e);
  ```
**3. These are the callbacks:**
  ```C
  static bool custom_ostream_callback(pb_ostream_t *stream, const pb_byte_t *buf, size_t count)
  {
      RINGBUFF_T *rb = (RINGBUFF_T *) stream->state;
      printf("custom_ostream_callback!-> count:%ld\n", count);
      return RingBuffer_InsertMult(rb, buf, count) == count;
  }
  
  static bool custom_field_encoding_callback(pb_ostream_t *stream, const pb_field_iter_t *field,
                                             void *const *arg)
  {
      printf("custom_field_encoding_callback!-> tag:%d, (char *) *arg:%s\n", field->tag,
             (char *) *arg);
  
      char *raw_string = (char *) *arg;
      if (!pb_encode_tag_for_field(stream, field))
      {
          return false;
      }
      return pb_encode_string(stream, (uint8_t *) raw_string, strlen(raw_string));
  }
  
  static bool custom_repeated_encoding_callback(pb_ostream_t *stream, const pb_field_iter_t *field,
                                                void *const *arg)
  {
      printf("custom_repeated_encoding_callback!-> tag:%d\n", field->tag);
  
      Batch_Sample *samples = (Batch_Sample *) *arg;
  
      for (uint16_t k = 0; k < 2; k++)
      {
          if (!pb_encode_tag_for_field(stream, field))
          {
              const char *error = PB_GET_ERROR(stream);
              printf("pb_encode error: %s\n", error);
              return false;
          }
          if (!pb_encode(stream, Batch_Sample_fields, &samples[k]))
          {
              const char *error = PB_GET_ERROR(stream);
              printf("pb_encode error: %s\n", error);
              return false;
          }
      }
      return true;
  }
  ```
**4. This is the output**
```log
custom_repeated_encoding_callback!-> tag:1
custom_ostream_callback!-> count:1
custom_field_encoding_callback!-> tag:2, (char *) *arg:channel_2000_200_1
custom_ostream_callback!-> count:1
custom_ostream_callback!-> count:1
custom_ostream_callback!-> count:18
custom_ostream_callback!-> count:1
custom_ostream_callback!-> count:2
custom_ostream_callback!-> count:1
custom_ostream_callback!-> count:8
custom_ostream_callback!-> count:1
custom_ostream_callback!-> count:8
custom_ostream_callback!-> count:1
custom_field_encoding_callback!-> tag:2, (char *) *arg:channel_2000_200_1
custom_ostream_callback!-> count:1
custom_ostream_callback!-> count:1
custom_ostream_callback!-> count:18
custom_ostream_callback!-> count:1
custom_ostream_callback!-> count:2
custom_ostream_callback!-> count:1
custom_ostream_callback!-> count:8
custom_ostream_callback!-> count:1
custom_ostream_callback!-> count:8
Encoded size: 84
```

PetteriAimonen, 2023-07-16:
What is the actual problem you are encountering?

If the resulting encoded data is not accepted by decoders, you can try e.g. https://protobuf-decoder.netlify.app/ to debug

TSMotter, 2023-07-16:
Thanks for the reply,

I was hoping that a more experienced eye could catch some basic error that I might be doing... 

For example, should I be calling `pb_encode` from within `custom_repeated_encoding_callback`? I feel like I'm encoding twice, since I'm calling `pb_encode` from `main()` also 

Regardless, I'll try the tool you suggested to debug.

TSMotter, 2023-07-16:
Yes, I just rubber ducked myself... 

It works now, it looks like I should be calling `pb_encode_submessage` instead of `pb_encode` within the callback

You debugging suggestion helped though (https://protobuf-decoder.netlify.app/)

Thank you

TSMotter, 2023-07-16:
Yes, I just rubber ducked myself...

It works now, it looks like I should be calling pb_encode_submessage instead of pb_encode within the callback

You debugging suggestion helped though (https://protobuf-decoder.netlify.app/)

Thank you

closed 2023-07-16 by TSMotter

PetteriAimonen, 2023-07-16:
Nice to hear you got it solved.

Indeed `pb_encode_submessage()` is needed - the difference is that it adds the length prefix needed for submessages. Not so easy to spot quickly in the code though :)