nanopb/nanopb #887: DeprecationWarning pkg_resources is deprecated as an API

fermentedfly, 2023-08-08:
Hi,

I recently switched to Python 3.11 and now get Deprecation warnings for `nanopb_generator.py:22` -> `import pkg_resources`

Regards,
Manuel

PetteriAimonen, 2023-08-08:
This was originally included for PyInstaller, but appears to be no longer required.

PetteriAimonen, 2023-11-11:
Fix published in nanopb-0.4.8.

idan-ben-ami, 2024-05-06:
I get the same warning using any of these:

generator/proto/__init__.py:import pkg_resources
generator/proto/__init__.py:        _builtin_proto_include = pkg_resources.resource_filename('grpc_tools', '_proto')
generator/proto/_utils.py:            import pkg_resources
generator/proto/_utils.py:            paths.append(pkg_resources.resource_filename('grpc_tools', '_proto'))

It is required when importing nanopb_pb2.py from the generated python script (myproto_pb2.py).

lubosz, 2024-07-19:
Because of this `nanopb_generator` fails to run on Python 3.12:

```
$ nanopb_generator
Traceback (most recent call last):
  File "/home/bmonkey/.local/bin/nanopb_generator", line 5, in <module>
    from nanopb.generator.nanopb_generator import main_cli
  File "/home/bmonkey/.local/share/pipx/venvs/nanopb/lib/python3.12/site-packages/nanopb/generator/nanopb_generator.py", line 57, in <module>
    from . import proto
  File "/home/bmonkey/.local/share/pipx/venvs/nanopb/lib/python3.12/site-packages/nanopb/generator/proto/__init__.py", line 10, in <module>
    import pkg_resources
ModuleNotFoundError: No module named 'pkg_resources'
```

jaskij, 2024-07-19:
To test for this, I modified `binary_packages.yml` to run using both Python 3.8 and whatever is the latest 3.x release (which at the time of writing is Python 3.12). To my surprise, [the job passes](https://github.com/nanopb/nanopb/actions/runs/10014462153/job/27684248087?pr=993).

@lubosz do you perhaps have a fresh install of a fast moving distro? Like Fedora?

lubosz, 2024-07-20:
@jaskij I do use Arch Linux, not a fresh install, but I can reproduce this in an virtualenv:
```
$ virtualenv fresh_env
created virtual environment CPython3.12.4.final.0-64 in 74ms
  creator CPython3Posix(dest=/fresh_env, clear=False, no_vcs_ignore=False, global=False)
  seeder FromAppData(download=False, pip=bundle, via=copy, app_data_dir=/.local/share/virtualenv)
    added seed packages: pip==24.1
  activators BashActivator,CShellActivator,FishActivator,NushellActivator,PowerShellActivator,PythonActivator

$ . fresh_env/bin/activate

(fresh_env) $ python --version
Python 3.12.4

(fresh_env) $ pip install nanopb==0.4.9.dev1481
Collecting nanopb==0.4.9.dev1481
  Downloading nanopb-0.4.9.dev1481-py2.py3-none-any.whl.metadata (5.8 kB)
Collecting protobuf>=3.19 (from nanopb==0.4.9.dev1481)
  Using cached protobuf-5.27.2-cp38-abi3-manylinux2014_x86_64.whl.metadata (592 bytes)
Downloading nanopb-0.4.9.dev1481-py2.py3-none-any.whl (53 kB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 53.9/53.9 kB 1.0 MB/s eta 0:00:00
Using cached protobuf-5.27.2-cp38-abi3-manylinux2014_x86_64.whl (309 kB)
Installing collected packages: protobuf, nanopb
Successfully installed nanopb-0.4.9.dev1481 protobuf-5.27.2

[notice] A new release of pip is available: 24.1 -> 24.1.2
[notice] To update, run: pip install --upgrade pip

(fresh_env) $ nanopb_generator
Traceback (most recent call last):
  File "/fresh_env/bin/nanopb_generator", line 5, in <module>
    from nanopb.generator.nanopb_generator import main_cli
  File "/fresh_env/lib/python3.12/site-packages/nanopb/generator/nanopb_generator.py", line 57, in <module>
    from . import proto
  File "/fresh_env/lib/python3.12/site-packages/nanopb/generator/proto/__init__.py", line 10, in <module>
    import pkg_resources
ModuleNotFoundError: No module named 'pkg_resources'
```

jaskij, 2024-07-21:
@PetteriAimonen I think this issue can be closed now?

closed 2024-09-19 by PetteriAimonen