nanopb/nanopb #889: Missing CMake option for PB_FIELD_32BIT ?

hpe-ykoehler, 2023-08-14:
I have a need for an app that requires PB_FIELD_32BIT to be set.  Since we used the nanopb as a static library that would imply we need to enable that would set the option and adjust the static library "public" definitions flags to also set the PB_FIELD_32BIT, does that make sense?

I can do the work to add the CMake option and the target_compile_definitions(), etc, but I am basically just trying to see if I miss something.

PetteriAimonen, 2023-10-19:
I think it would be excessive to have CMake wrappers for each build option nanopb has. There is probably some other way to directly specify compilation options for CMake, but I don't know how it works.

closed 2023-10-19 by PetteriAimonen

hpe-ykoehler, 2023-10-19:
I have been playing on this and create a patch to set:
```
+        target_compile_definitions(protobuf-nanopb PUBLIC
+            PB_FIELD_32BIT
         )
```

when doing such a change when someone include nanopb target as a link element, the flags automatically move to the application linking with the library as well, so the application would do something like:

```
target_link_libraries(myapp PRIVATE nanopb::protobuf-nanopb)
```

And that would ensure the app aldo get the -DPB_FIELD_32BIT added.

Next thing I am missing is having this flag as an option in the conan recipe, so that one can turn it on in a conan profile, so that it gets set in cmake and then gets set in the gcc flags chain of application linking with nanopb.

If you do not want to spend time on this, will you be open to receive a PR about it once I am completed?

hpe-ykoehler, 2023-10-19:
I would not implement all flags, but the work would give the road for people to add more flags as they need basically.

PetteriAimonen, 2023-10-19:
I'm not against it in principle. For builds on high-powered systems (where CMake & Conan would be more common as build systems), PB_FIELD_32BIT might even be a sensible default option.