nanopb/nanopb #895: Why inner message becomes optional

nikitos1550, 2023-08-20:
Hello! I am learning nanopb (and protobufs in general) and faced strange thing.
I have message that has one field as another message and I expect it to be required, but it is generated as optional.
I cannot understan why, could somebody explain me?

Here is my proto file:
```
syntax = "proto3";

//import "nanopb.proto";
//import "google/protobuf/timestamp.proto";

option go_package = ".";

message Timestamp {
  uint32 seconds = 1;
  uint32 nanos = 2;
}

message INAReport {
        Timestamp ts = 1;
        uint32 channel = 2;
        optional uint32 period = 3;
        uint32 value = 4;
        //repeated uint32 value = 4 [(nanopb).max_count = 10];
}

message McuReport {
        oneof payload {
                INAReport inareport = 1;
        }
}
```
For some reason Timestamp is optional.
```
/* Struct definitions */
typedef struct _Timestamp {
    uint32_t seconds;
    uint32_t nanos;
} Timestamp;

typedef struct _INAReport {
    bool has_ts;   // <--------- why this ?
    Timestamp ts;
    uint32_t channel;
    bool has_period;
    uint32_t period;
    uint32_t value; /* repeated uint32 value = 4 [(nanopb).max_count = 10]; */
} INAReport;

typedef struct _McuReport {
    pb_size_t which_payload;
    union {
        INAReport inareport;
    } payload;
} McuReport;
```

This is how I generate C code:
```
python3 ../3rdparty/nanopb/generator/nanopb_generator.py \
        --protoc-opt=--experimental_allow_proto3_optional \
        simple.proto
```

apcountryman, 2023-08-20:
proto 3 has no concept of required fields. If a label is not provided for a proto 3 field, [it has the default "implicit field presence" label](https://protobuf.dev/programming-guides/proto3/#field-labels). If the field is a singular message, [its presence is tracked explicitly](https://protobuf.dev/programming-guides/field_presence/#presence-in-proto3-apis).

nikitos1550, 2023-08-20:
> proto 3 has no concept of required fields. If a label is not provided for a proto 3 field, [it has the default "implicit field presence" [...]

I see, thanks.

closed 2023-08-20 by nikitos1550