nanopb/nanopb #896: PROPOSAL: add const void* arg to pb_callback_t

Xeenych, 2023-08-21:
Please add `const void* const_arg` to pb_callback_t struct and use it in `encode` function for better const correctness and C++ compatibility

```
struct pb_callback_s {
    /* Callback functions receive a pointer to the arg field.
     * You can access the value of the field as *arg, and modify it if needed.
     */
    union {
        bool (*decode)(pb_istream_t *stream, const pb_field_t *field, void **arg);
        bool (*encode)(pb_ostream_t *stream, const pb_field_t *field, void * const *arg);
    } funcs;

    /* Free arg for use by callback */
  union {
  void *arg;
  const void* const_arg;
  }
};
```

PetteriAimonen, 2023-08-21:
I'm not sure about this, especially for C++ it seems like an explicit const_cast<> would be better than hiding it this way.

Xeenych, 2023-08-21:
this results in ugly  `.arg = const_cast<void*>(static_cast<const void*>(parameter))`

closed 2023-11-11 by PetteriAimonen