nanopb/nanopb #900: Support for Editions

brentru, 2023-09-15:
Hi @PetteriAimonen  - are you planning to support "Protobuf Editions" as described in https://protobuf.dev/news/2023-06-29 within nanopb? A tool, prototiller, is being provided as a way to migrate the protobuf files themselves (https://protobuf.dev/editions/overview/#prototiller). Are the converted .proto files going to work with nanopb?

PetteriAimonen, 2023-09-15:
I have no idea - it seems there is pretty little information available, especially on how the "editions" will affect the output format from `protoc` to language generators.

brentru, 2023-09-18:
Thanks for the insight, we'll take a "wait and see" approach. 

Feel free to close out this issue, or keep it open if you feel "editions" will need a correlated issue.

PetteriAimonen, 2023-09-18:
My expectation is that most of the changes will be handled by protoc and are not be visible to nanopb generator.

But yeah, if anyone finds specific issues, feel free to reopen or add a separate issue.

mkruskal-google, 2024-10-09:
Hey Petteri,

We do have some documentation on this now: https://protobuf.dev/editions/implementation/.    Protoc does handle large parts of the work, but at least your generator (which doesn't use our C++ plugin infrastructure) will need some additional support.

PetteriAimonen, 2026-03-30:
@EliasStar Probably after I get 1.0 out unless something urgent comes along.. next two months are looking busy with work stuff :/