nanopb/nanopb #903: Option submsg_callback not working

ottmar-zittlau, 2023-09-30:
Hi,

I want to use a protobuf using the `submsg_callback`-flag together with an oneof-field:
```
message data {
   option (nanopb_msgopt).submsg_callback = true;
   oneof val {
      bool bool_val = 1;
      uint32 uint8_val = 2;
      uint32 uint16_val = 3
      }
}
```
I'm currently trying to update nanopb from 0.4.4 to 0.4.7 - and it seems that the `submsg_callback`-flag doesn't work anymore.

I debugged the generator script and realized that in the constructor of the `OneOf`-class - the `has_msg_cb`-flag is set to `false`:
```
self.has_msg_cb = False
```
if I set it to
```
self.has_msg_cb = oneof_options.submsg_callback
```
everything seems to work.

I also noticed that this flag is set in the `add_field`-method of `OneOf` if the field's pbtype is `MSG_W_CB`. This seems to be set in the constructor of the `Field`-class - but, if I understand it correctly - only for fields of message type - so not for the simple bool, uint etc. stuff, I'm using in the oneof. Do I have to wrap the oneof in a separate message?

Thanks in advance and best regards
oz

PetteriAimonen, 2023-09-30:
I think it is intentional that it is only called for submessages. I'm not sure if that was a good choice, though.

ottmar-zittlau, 2023-09-30:
Thanks for the quick reply!

In the meantime, I noticed that the version of the protobuf-library that is loaded (as displayed with the verbose flag) and the protobuf compiler don't match.
I probably have to figure this out first.

Best regards,
oz

closed 2023-09-30 by ottmar-zittlau