nanopb/nanopb #909: CMake find package config mode doesn't include the CMake macro nanopb_generate_cpp() and the MODULE mode does not include the targets

hpe-ykoehler, 2023-10-19:
I have been playing with nanopb and cmake, and there is a problem in my view as to how the package provide the FindNanopb.cmake file and such.

I see for example that nanopb provides nanopb:: namespace and targets but there is not example provided where this is use.  To use those targets normally one creates an application and use.

```
find_package(nanopb CONFIG)

target_link_libraries(myapp PRIVATE nanopb:: <target>)
```

Yet, doing so cause an issue because then the developer cannot use the CMake macro nanopb_generate_cpp(), which only gets added when adding nanopb using the MODULE mode:

find_package(nanopb MODULE) vs find_package(nanopb CONFIG)

My understanding is that the macro nanopb_generate_cpp() should be in its own .cmake file such as nanopb_macros.cmake and this file should then be included by both FindNanopb.cmake and nanopb-config.cmake, as to make it available in both mode.

The use of CMake targets is related to what they call modern CMake: https://cliutils.gitlab.io/modern-cmake/

The MODULE mode is the old way, where one would have to use something like

```
target_include_directories(myapp PRIVATE NANOPB_INCLUDES)
target_link_libraries(myapp PRIVATE NANOPB_LIBRARIES)
```

and then have to ensure out-of-band that the same gcc DEFINES flags are used across the compilation chain using that library, this is where the modern CMake shines in that it takes care of the gcc related details for the developers.

hpe-ykoehler, 2023-10-19:
If one were to write a sample app using CONFIG mode and the macro:
```
find_package(nanopb CONFIG)

nanopb_generate_cpp(PROTO_SRC PROTO_HEADER simple.proto)

add_executable(${PROJET_NAME} simple.c ${PROTO_SRC} ${PROTO_HEADER})

target_link_libraries(simple PRIVATE nanopb:: <target>)
```

CMake would then report an error.

PetteriAimonen, 2023-10-19:
I must admit that my CMake knowledge is not enough to understand this.

PetteriAimonen, 2023-10-27:
I think #911 may also be tangentially related to this problem.