nanopb/nanopb #910: Feature Request: Support App Privacy Manifest

MichaelFoden, 2023-10-25:
Apple required App Privacy manifests until Spring 2024 .
If using particular APIs as outlined by Apple, third-party SDKs also require support.
I would like to know when you adopt this change.
If no action is required, please let us know.

●Questions

Dose Flanimatedimage sdk use those particular APIs?
If it’s used, can support be provided for it?
●References
https://developer.apple.com/videos/play/wwdc2023/10060/
https://developer.apple.com/news/?id=av1nevon
https://developer.apple.com/app-store/user-privacy-and-data-use/

Please check it out. Thanks!

PetteriAimonen, 2023-10-25:
This does not seem relevant to the nanopb core project. If you are using some project that packages nanopb for use on iOS, please contact them instead.

MichaelFoden, 2023-10-25:
Thank you for quick response.
Does that mean nanopb doesn’t use any of Apple’s APIs?

esteluk, 2023-12-12:
@PetteriAimonen I don't know if you've seen https://developer.apple.com/support/third-party-SDK-requirements/ yet, which explicitly calls out nanopb (and/or its consumers) as requiring a mandatory manifest.

paulb777, 2023-12-12:
We're working on privacy manifests for Firebase now and should be able to do one for nanopb (at least 0.3.*), as well, in the next several weeks. https://github.com/firebase/firebase-ios-sdk/issues/11490  cc: @ncooke3

nabs-m, 2023-12-19:
Can we reopen this issue since there seems to be a plan to add a privacy manifest to nanopb?

PetteriAimonen, 2023-12-20:
We can reopen it, but I reiterate that I do not have a plan nor time to study what Apple is trying to require. If Paul can or anyone else provides a pull request, I'll of course review and merge it as appropriate.

PetteriAimonen, 2024-02-16:
This should now be resolved thanks to ncooke3.

PR #938 (2024-02-16): [nanopb] Add Apple privacy manifest
Add a privacy manifest for nanopb. The library does not use any ["required reason API"](https://developer.apple.com/documentation/bundleresources/privacy_manifest_files/describing_use_of_required_reason_api?language=objc).

Proposing to:
- merge this in the `maintenance_0.3` branch
- publish a `0.3.9.10`  tag after merging

Fixes #910

cc: @paulb777
closed 2024-02-16 by PetteriAimonen