nanopb/nanopb #911: Compatibility of FindNanopb.cmake and CMakeLists.txt installation

PetteriAimonen, 2023-10-27:
While doing #907, I noticed that CMakeLists.txt installation currently doesn't install FindNanopb.cmake, and the two don't really co-operate at all.

For ease of use in various configurations, I think these changes would be needed:

1. Nanopb CMake installation should expose any relevant paths in the installed `nanopb-config.cmake` file.
2. FindNanopb.cmake should be able to find a system-wide nanopb installation.
3. Nanopb CMake installation should install FindNanopb.cmake to somewhere that cmake can find it without having the nanopb source directory available.

rhaberkorn, 2024-11-17:
Currently when using FindNanopb.cmake (e.g. `find_package(Nanopb REQUIRED)`), you need to specify the location of the entire source tree in `NANOPB_SRC_ROOT_FOLDER`, even though a system-wide installation installs a static library and headers. It's not meant to be used that way?

PetteriAimonen, 2024-11-17:
@rhaberkorn So far the CMakeLists.txt and FindNanopb.cmake have been written by different contributors, so I guess little thought has been given to their interoperability. I think it would make sense to have them co-operate.

akohlsmith, 2025-12-29:
it sounds like this is still an issue in December 2025. You can use CMake to find nanopb, but you can't run the generator because it tries to run it in the system "namespace" not the cmake project's namespace. I've seen attempts to work around this by creating a venv inside of the cmake project with creative use of macros and variable settings, but they all appear to fail because even though cmake builds the venv and uses the venv to run the python generator, the generator still seems to want protobufs to be installed on the host. And let's not get started on the state of trying to have multiple python utilities installed in the system.

Is there not a clean, self-contained way to include nanopb in a cmake project that does not demand manual installation of a project venv? This is completely contrary to the concept that cmake tries to solve and which nanopb appears to embrace.