nanopb/nanopb #914: Stream decoding compressed file

JAremko, 2023-11-02:
Given constrained RAM is it possible to use [`heatshrink`](https://github.com/atomicobject/heatshrink) or some other compression library to stream decode a proto 3 payload that was compressed?
 By `Stream decoding` I mean some kind of an integration with a compression library that allows decoding data without decompressing the whole payload.

MaJerle, 2023-11-03:
You can use input stream with callbacks, that feeds the parser byte by byte, and you do partial decoding in the callback. I guess it is a perfect solution for you.

PetteriAimonen, 2023-11-03:
Yeah, something like this could work (untested code):

```
struct hsd_stream_t
{
    heatshrink_decoder *hsd;
    FILE *file;
    uint8_t *tmpbuf;
    size_t tmpbuf_size;
};

static bool read_callback(pb_istream_t *stream, uint8_t *buf, size_t count)
{
    struct hsd_stream_t *state = (struct hsd_stream_t*)(stream->state);
    size_t num_read = 0;
    size_t result = 0;
    
    while (num_read < count)
    {
        heatshrink_decoder_poll(state->hsd, buf + num_read, count - num_read, &result);
        num_read += result;
        
        if (num_read < count)
        {
            // Get more data from file
            result = fread(state->file, 1, state->tmpbuf_size, state->tmpbuf);
            heatshrink_decoder_sink(state->hsd, state->tmpbuf, result, &result);
            
            // TODO: Handle EOF and error conditions
        }
    }
    
    return true;
}
```

[examples/network_server/common.c](examples/network_server/common.c) uses callbacks to bind stream into a network socket, so it can be useful reference for implementing your own stream. Also see [docs](https://jpa.kapsi.fi/nanopb/docs/concepts.html#streams).

closed 2023-11-03 by JAremko