nanopb/nanopb #915: FeatReq: Add option macro for user fields and include of user header file

MaJerle, 2023-11-03:
I find nanopb an outstanding library, well done by original author and all the contributors. I'd like to propose a new feature, namely custom field in the generated structure from the `.proto` file.

Consider this `.proto` file:

```
syntax = "proto3";

/* e is repeated, but data will not come in packed */
message Test4 {
  optional string d = 4;
  repeated int32 e = 5;
}

/* Test of packed */
message Test5 {
  repeated int32 f = 6;
}

message Test6 {
    repeated int32 g = 7;
}
```

Options file
```
Test4.d max_size:13
Test4.e max_count:16
Test5.f max_count:16
```

The message `g` is repeated, but has no defined max count. This means that callback system is used, that allows user to parse the object and do with the data whatever he wants.

An idea is to add custom fields to the structure, primary use would be to allow custom memory allocation when decoding the message. User is then responsible to allocate memory in the callback, parse the data, and assign the values to the structure.

Sometimes application doesn't want to allocate fixed max amount in advance since total structure may become too big. Instead, it may dynamically allocate memory only for the fields that are actually present in the message.

Proposal is to generate output like that:

- Add macro `PB_INCLUDE_USER_OPTS` which, when exists, includes file `pb_user.h`, that is primary used to allow user to add custom fields outside of generated header file
- Add not-defined check for user fields for each structure
- Add user macro to the structure parameter

User custom header:
```c
#ifndef PB_USER_HDR_H
#define PB_USER_HDR_H

#define STRUCT_TEST6_USER_FIELDS                                                                                       \
    int32_t* g_arr;                                                                                                    \
    size_t g_arr_count;

#endif
```

Generated header:
```c
/* Automatically generated nanopb header */
/* Generated by nanopb-0.4.8-dev */

#ifndef PB_MESSAGE_PB_H_INCLUDED
#define PB_MESSAGE_PB_H_INCLUDED
#include <pb.h>

#if PB_PROTO_HEADER_VERSION != 40
#error Regenerate this file with the current version of nanopb generator.
#endif

/* Include user file if defined in the compiler settings */
#ifdef PB_INCLUDE_USER_OPTS
#include "pb_user.h"
#endif

/* Define if not set by user */
#ifndef STRUCT_TEST4_USER_FIELDS
#define STRUCT_TEST4_USER_FIELDS
#endif
#ifndef STRUCT_TEST5_USER_FIELDS
#define STRUCT_TEST5_USER_FIELDS
#endif
#ifndef STRUCT_TEST6_USER_FIELDS
#define STRUCT_TEST6_USER_FIELDS
#endif

/* Struct definitions */
/* e is repeated, but data will not come in packed */
typedef struct _Test4 {
    bool has_d;
    char d[13];
    pb_size_t e_count;
    int32_t e[16];

    /* User fields */
    STRUCT_TEST4_USER_FIELDS
} Test4;

/* Test of packed */
typedef struct _Test5 {
    pb_size_t f_count;
    int32_t f[16];

    /* User fields */
    STRUCT_TEST5_USER_FIELDS
} Test5;

typedef struct _Test6 {
    pb_callback_t g;

    /* User fields */
    STRUCT_TEST6_USER_FIELDS
} Test6;

//...the rest afterwards

```

Does it many sense?

PetteriAimonen, 2023-11-03:
Your idea makes sense, but it also overlaps with some existing functionality.

Primarily, I would suggest using `(nanopb).callback_datatype`.
It can be set to a structure:

    message Test6 {
        repeated int32 g = 7 [(nanopb).callback_datatype = 'struct { int32_t *arr; size_t arr_count; }'];
    }

This automatically enables [name-bound callbacks](https://jpa.kapsi.fi/nanopb/docs/whats_new.html#callbacks-bound-by-function-name) so you don't have to set a function pointer. Just define a function by name `Test6_callback()`. The prototype declaration of that function will be in the generated `.pb.h` file. Callback behavior remains the same, in fact the `pb_callback_t` function pointers are handled by name-bound [pb_default_field_callback()](https://github.com/nanopb/nanopb/blob/master/pb_common.c#L300).

---

One can also take advantage of nesting structs in C:

     struct MyTest6 {
            Test6 message;
            int32_t* arr;
            size_t arr_count;
     };

Now a pointer to MyTest6 is also a pointer to Test6. This equivalency is called "pointer to first member" in C standard. You can call `pb_decode()` with the Test6 pointer. In a callback, you can cast `field->message` to `(MyTest6*)` to access the extra fields.

---

Finally, there is the more extensive way to [customize structures using `PB_BIND()`](https://jpa.kapsi.fi/nanopb/docs/whats_new.html#binding-message-types-to-custom-structures). That allows binding the generated protobuf definitions to an entirely custom message struct, as long as the field names match.

PetteriAimonen, 2023-11-03:
Oh, and forgot to mention about the including user header file. That can be done with file-level option:

     option (nanopb_fileopt).include = 'myheader.h';

After that the types from there can be used in `callback_datatype`.

MaJerle, 2023-11-03:
Ou, thanks. Points taken.

closed 2023-11-03 by MaJerle