nanopb/nanopb #920: "Could not determine size for submessage" when file uses (nanopb_fileopt).package

canton7, 2023-11-10:
To repro, create two files:

### first.proto

```proto
syntax = "proto3";

package org.first;

import "nanopb.proto";
import "second.proto";

option (nanopb_fileopt).package = "first";

message First {
    org.second.Second second = 1;
}
```

### second.proto

```proto
syntax = "proto3";

package org.second;

message Second {
}
```

When you run nanopb_generator:

```
nanopb_generator Protobufs/first.proto -C --strip-path -D Generated -I Protobufs
Could not determine size for submessage first_org_second_Second, using default 256
Could not determine size for submessage first_org_second_Second, using default 256
Could not determine size for submessage first_org_second_Second, using default 256
```

This only happens when `option (nanopb_fileopt).package` is used in `first.proto`: using it in `second.proto`

The name `first_org_second_Second` in the error message looks particularly mangled?

PetteriAimonen, 2023-11-10:
I don't seem to be getting this error on newest master, though I do get it on 0.4.7. I think the issue has been fixed by pull request #899.

PetteriAimonen, 2023-11-11:
Fix published in nanopb-0.4.8.

closed 2023-11-11 by PetteriAimonen

canton7, 2023-11-11:
Fantastic, thank you so much!