nanopb/nanopb #921: message pre-checks?

amery, 2023-11-10:
hi, what's the correct way to see if we have enough data on the stream to decode a message?

decode returning a boolean doesn't let one distinguish between:
1) incomplete
2) invalid
3) wrong message type

which could need to be handled differently

PetteriAimonen, 2023-11-10:
Protobuf as a format does not allow precise detection of any of these conditions.
In fact, empty message (of 0 bytes) is a valid encoding for any proto3 message type.

To verify the integrity of a message, external means are needed (like a checksum or a signature).

amery, 2023-11-10:
> Protobuf as a format does not allow precise detection of any of these conditions. In fact, empty message (of 0 bytes) is a valid encoding for any proto3 message type.
> 
> To verify the integrity of a message, external means are needed (like a checksum or a signature).

not even using tag+length to determine if we need more bytes?

PetteriAimonen, 2023-11-10:
Only if the message gets cut in middle of a field. But because protobuf messages are not self-delimiting, you must already have some method of knowing when the message ends.

stream->errmsg will give a more detailed reason for an error return, but in general this is meant only for debugging purposes. I try to keep the messages stable, so it's not terribly wrong to `strcmp()` with e.g. "end-of-stream".

You can perform sort-of validation without decoding by repeatedly calling `pb_decode_tag()` and `pb_skip_field()`, but that will only detect some errors and not all, as many corruptions and truncations of protobuf messages result in a valid message.

amery, 2023-11-10:
my primary goal is to know when I have enough data to _pb_decode_ or disconnect someone sending nonsense data.
checking `stream->errmsg` seems like a fair _work-around_ to answer if it's **incomplete** and if it's **definitively not** the kind of wrapper message we expect

PetteriAimonen, 2023-11-11:
I don't see a clear use case where a best-effort distinction between "truncated message" and "otherwise corrupted message" would be useful.

closed 2023-11-11 by PetteriAimonen

amery, 2023-11-11:
> I don't see a clear use case where a best-effort distinction between "truncated message" and "otherwise corrupted message" would be useful.

It's the case of asynchronous io, you are notified there Is something you read, and you read it into a buffer.
Now you need to decide if you process it or wait. if pb_decode succeeds, all good. But If it doesn't, what do you do? keep appending and retrying. but if it was the wrong type of message or complete non-sense, you'll append data for ever

apcountryman, 2023-11-11:
As previously mentioned, [the Protocol Buffer wire format is not self delimiting](https://protobuf.dev/programming-guides/techniques/#streaming) so the data stream needs to include a message delimiting mechanism so that you can know when you have received a complete encoded message. A common method for doing this is prefixing an encoded message with the size of the encoded message (using `pb_encode_ex()` with the `PB_ENCODE_DELIMITED` flag and `pb_decode_ex()` with the `PB_DECODE_NULLTERMINATED` flag is one way of doing this).