nanopb/nanopb #923: nanopb 0.4.8 generates pb_callback_t for fields that specify a max size

jmecom, 2023-11-15:
**Steps to reproduce the issue**

Example input:

```
syntax = "proto3";

package demo;

import "nanopb.proto";

message foo {
  bytes bar = 1 [(nanopb).max_size = 1024];
}

message test {
  oneof msg {
    foo foo = 1;
  }
}
```

Compile with:
```
nanopb_generator -I. --output-dir=/tmp/example.proto
```

**What happens?**

On version 0.4.7, the compiled types look like this:

```
/* Struct definitions */
typedef PB_BYTES_ARRAY_T(1024) demo_foo_bar_t;
typedef struct _demo_foo {
    demo_foo_bar_t bar;
} demo_foo;

typedef struct _demo_test {
    pb_size_t which_msg;
    union {
        demo_foo foo;
    } msg;
} demo_test;
```

But on 0.4.8, they're like this:
```
/* Struct definitions */
typedef struct _demo_foo {
    pb_callback_t bar;
} demo_foo;

typedef struct _demo_test {
    pb_size_t which_msg;
    union {
        demo_foo foo;
    } msg;
} demo_test;
```

Full headers below.

### 0.4.7
```
/* Automatically generated nanopb header */
/* Generated by nanopb-0.4.7 */

#ifndef PB_DEMO_EXAMPLE_PB_H_INCLUDED
#define PB_DEMO_EXAMPLE_PB_H_INCLUDED
#include <pb.h>

#if PB_PROTO_HEADER_VERSION != 40
#error Regenerate this file with the current version of nanopb generator.
#endif

/* Struct definitions */
typedef PB_BYTES_ARRAY_T(1024) demo_foo_bar_t;
typedef struct _demo_foo {
    demo_foo_bar_t bar;
} demo_foo;

typedef struct _demo_test {
    pb_size_t which_msg;
    union {
        demo_foo foo;
    } msg;
} demo_test;

#ifdef __cplusplus
extern "C" {
#endif

/* Initializer values for message structs */
#define demo_foo_init_default                    {{0, {0}}}
#define demo_test_init_default                   {0, {demo_foo_init_default}}
#define demo_foo_init_zero                       {{0, {0}}}
#define demo_test_init_zero                      {0, {demo_foo_init_zero}}

/* Field tags (for use in manual encoding/decoding) */
#define demo_foo_bar_tag                         1
#define demo_test_foo_tag                        1

/* Struct field encoding specification for nanopb */
#define demo_foo_FIELDLIST(X, a) \
X(a, STATIC,   SINGULAR, BYTES,    bar,               1)
#define demo_foo_CALLBACK NULL
#define demo_foo_DEFAULT NULL

#define demo_test_FIELDLIST(X, a) \
X(a, STATIC,   ONEOF,    MESSAGE,  (msg,foo,msg.foo),   1)
#define demo_test_CALLBACK NULL
#define demo_test_DEFAULT NULL
#define demo_test_msg_foo_MSGTYPE demo_foo

extern const pb_msgdesc_t demo_foo_msg;
extern const pb_msgdesc_t demo_test_msg;

/* Defines for backwards compatibility with code written before nanopb-0.4.0 */
#define demo_foo_fields &demo_foo_msg
#define demo_test_fields &demo_test_msg

/* Maximum encoded size of messages (where known) */
#define demo_foo_size                            1027
#define demo_test_size                           1030

#ifdef __cplusplus
} /* extern "C" */
#endif

#endif
```

### 0.4.8
```
/* Automatically generated nanopb header */
/* Generated by nanopb-0.4.8 */

#ifndef PB_DEMO_EXAMPLE_PB_H_INCLUDED
#define PB_DEMO_EXAMPLE_PB_H_INCLUDED
#include <pb.h>

#if PB_PROTO_HEADER_VERSION != 40
#error Regenerate this file with the current version of nanopb generator.
#endif

/* Struct definitions */
typedef struct _demo_foo {
    pb_callback_t bar;
} demo_foo;

typedef struct _demo_test {
    pb_size_t which_msg;
    union {
        demo_foo foo;
    } msg;
} demo_test;

#ifdef __cplusplus
extern "C" {
#endif

/* Initializer values for message structs */
#define demo_foo_init_default                    {{{NULL}, NULL}}
#define demo_test_init_default                   {0, {demo_foo_init_default}}
#define demo_foo_init_zero                       {{{NULL}, NULL}}
#define demo_test_init_zero                      {0, {demo_foo_init_zero}}

/* Field tags (for use in manual encoding/decoding) */
#define demo_foo_bar_tag                         1
#define demo_test_foo_tag                        1

/* Struct field encoding specification for nanopb */
#define demo_foo_FIELDLIST(X, a) \
X(a, CALLBACK, SINGULAR, BYTES,    bar,               1)
#define demo_foo_CALLBACK pb_default_field_callback
#define demo_foo_DEFAULT NULL

#define demo_test_FIELDLIST(X, a) \
X(a, STATIC,   ONEOF,    MESSAGE,  (msg,foo,msg.foo),   1)
#define demo_test_CALLBACK NULL
#define demo_test_DEFAULT NULL
#define demo_test_msg_foo_MSGTYPE demo_foo

extern const pb_msgdesc_t demo_foo_msg;
extern const pb_msgdesc_t demo_test_msg;

/* Defines for backwards compatibility with code written before nanopb-0.4.0 */
#define demo_foo_fields &demo_foo_msg
#define demo_test_fields &demo_test_msg

/* Maximum encoded size of messages (where known) */
/* demo_foo_size depends on runtime parameters */
/* demo_test_size depends on runtime parameters */

#ifdef __cplusplus
} /* extern "C" */
#endif

#endif
```

PetteriAimonen, 2023-11-15:
Hmm, I'm getting the expected `PB_BYTES_ARRAY_T()` even with 0.4.8.
I wonder if there is something going wrong with the loading of either `nanopb.proto` or `nanopb_pb2.py`.

Could you include the following information:

* Output messages when you add `--verbose` to your current call to nanopb_generator
* Version info with `nanopb_generator --verbose --version`
* Whether nanopb_generator is installed system-wide or run from the source directory.
* Whether specifying `-s max_size:123` on the command line makes any difference to the output `.pb.h` file.

jmecom, 2023-11-15:
### nanopb_generator --verbose

0.4.7
```
Nanopb version nanopb-0.4.7
Google Python protobuf library imported from .hermit/python/lib/python3.10/site-packages/google/protobuf/__init__.py, version 4.23.4
Options file not found: google/protobuf/descriptor.options
Options for google/protobuf/descriptor.proto: 
Options for google.protobuf.FileDescriptorSet: 
Options for google.protobuf.FileDescriptorSet.file: 
Options for google.protobuf.FileDescriptorProto: 
Options for google.protobuf.FileDescriptorProto.name: 
Options for google.protobuf.FileDescriptorProto.package: 
Options for google.protobuf.FileDescriptorProto.dependency: 
Options for google.protobuf.FileDescriptorProto.public_dependency: 
Options for google.protobuf.FileDescriptorProto.weak_dependency: 
Options for google.protobuf.FileDescriptorProto.message_type: 
Options for google.protobuf.FileDescriptorProto.enum_type: 
Options for google.protobuf.FileDescriptorProto.service: 
Options for google.protobuf.FileDescriptorProto.extension: 
Options for google.protobuf.FileDescriptorProto.options: 
Options for google.protobuf.FileDescriptorProto.source_code_info: 
Options for google.protobuf.FileDescriptorProto.syntax: 
Options for google.protobuf.DescriptorProto: 
Options for google.protobuf.DescriptorProto.name: 
Options for google.protobuf.DescriptorProto.field: 
Options for google.protobuf.DescriptorProto.extension: 
Options for google.protobuf.DescriptorProto.nested_type: 
Options for google.protobuf.DescriptorProto.enum_type: 
Options for google.protobuf.DescriptorProto.extension_range: 
Options for google.protobuf.DescriptorProto.oneof_decl: 
Options for google.protobuf.DescriptorProto.options: 
Options for google.protobuf.DescriptorProto.reserved_range: 
Options for google.protobuf.DescriptorProto.reserved_name: 
Options for google.protobuf.DescriptorProto.ExtensionRange: 
Options for google.protobuf.DescriptorProto.ExtensionRange.start: 
Options for google.protobuf.DescriptorProto.ExtensionRange.end: 
Options for google.protobuf.DescriptorProto.ExtensionRange.options: 
Options for google.protobuf.DescriptorProto.ReservedRange: 
Options for google.protobuf.DescriptorProto.ReservedRange.start: 
Options for google.protobuf.DescriptorProto.ReservedRange.end: 
Options for google.protobuf.ExtensionRangeOptions: 
Options for google.protobuf.ExtensionRangeOptions.uninterpreted_option: 
Options for google.protobuf.ExtensionRangeOptions.extensions: 
Options for google.protobuf.FieldDescriptorProto: 
Options for google.protobuf.FieldDescriptorProto.name: 
Options for google.protobuf.FieldDescriptorProto.number: 
Options for google.protobuf.FieldDescriptorProto.label: 
Options for google.protobuf.FieldDescriptorProto.type: 
Options for google.protobuf.FieldDescriptorProto.type_name: 
Options for google.protobuf.FieldDescriptorProto.extendee: 
Options for google.protobuf.FieldDescriptorProto.default_value: 
Options for google.protobuf.FieldDescriptorProto.oneof_index: 
Options for google.protobuf.FieldDescriptorProto.json_name: 
Options for google.protobuf.FieldDescriptorProto.options: 
Options for google.protobuf.FieldDescriptorProto.Type: 
Options for google.protobuf.FieldDescriptorProto.Label: 
Options for google.protobuf.OneofDescriptorProto: 
Options for google.protobuf.OneofDescriptorProto.name: 
Options for google.protobuf.OneofDescriptorProto.options: 
Options for google.protobuf.EnumDescriptorProto: 
Options for google.protobuf.EnumDescriptorProto.name: 
Options for google.protobuf.EnumDescriptorProto.value: 
Options for google.protobuf.EnumDescriptorProto.options: 
Options for google.protobuf.EnumDescriptorProto.reserved_range: 
Options for google.protobuf.EnumDescriptorProto.reserved_name: 
Options for google.protobuf.EnumDescriptorProto.EnumReservedRange: 
Options for google.protobuf.EnumDescriptorProto.EnumReservedRange.start: 
Options for google.protobuf.EnumDescriptorProto.EnumReservedRange.end: 
Options for google.protobuf.EnumValueDescriptorProto: 
[... 404 lines cut ...]
Options for demo.test: proto3: true

Options for demo.test.msg: proto3: true

Options for demo.test.foo: proto3: true

Writing to /tmp/new/example.pb.h and /tmp/new/example.pb.c
```

### version info
```
❯❯❯ nanopb_generator --verbose --version
nanopb-0.4.7
```

```
❯❯❯ nanopb_generator --version
nanopb-0.4.8
```

nanopb_generator is on my path, but is installed 'locally' here: `.hermit/python/bin/nanopb_generator`. I'm managing it with [Hermit](https://cashapp.github.io/hermit/)

### passing max_size

Yes, this works; the resulting type is a `PB_BYTES_ARRAY_T(123)`

PetteriAimonen, 2023-11-15:
For some reason I still cannot reproduce the problem.

You could try ruling out any installation issues or lingering old packages by doing this test in a clean virtual env:

     python3 -m venv test_env
     test_env/bin/pip install protobuf grpcio-tools
     test_env/bin/python ...../nanopb/generator/nanopb_generator.py test.proto

jmecom, 2023-11-20:
Hmm, okay. Haven't looked into this further, but I'll close since it seems like my environment is the issue. Thanks!

closed 2023-11-20 by jmecom

drorgl, 2024-04-13:
If anyone else encounters this problem, this worked for me:
```
pip install --upgrade protobuf grpcio-tools
```