nanopb/nanopb #930: PB_BIND error

Libong, 2023-12-25:
When I add the automatically generated xx. pbh file to Keil, the PB_ BIND in the file compilation error message will prompt as follows

Explicit type is missing ("int" accumulated)

Expected a type specification

Expected a "{"

I want to ask how to modify it

The file content is
PB_BIND(xxx, xxx, AUTO)

PetteriAimonen, 2023-12-25:
You'll need to get more information from your compiler about the error, and try to minimize the .proto file to minimal version that reproduces the error. Preprocessing the file first and compiling separately may give a more specific line number for the problem.

closed 2023-12-28 by Libong

Libong, 2023-12-28:
When I define PB_ NO_ STATIC_ASSERT in the PB. h file , it can be compiled without any errors