nanopb/nanopb #932: Add encoded size to C++ output

denravonska, 2024-01-11:
We have a message queue where we can push messages of different types to be funneled further by a task. The message handling is a lot of repetition based on the data type

```Cpp
case SendDisconnect:
    PublishMessageFromBuffer<Disconnect, Disconnect_size>(disconnectTopic, Disconnect_fields);
    break;
```

This is both tedious to read and error prone if you specify the wrong size (`Connect_size` vs `Disconnect_size`). 
Right now we're using the C interface and are considering switching to the C++ helpers. If the helpers also included the maximum size those calls could be made easier to read and harder to get wrong:

```Cpp
case SendDisconnect:
    PublishMessageFromBuffer<Disconnect>(disconnectTopic);
    break;
```

PetteriAimonen, 2024-01-11:
Seems reasonable. I'm not going to work on this myself any time soon, but it would probably be quite straightforward to imitate pull request #931 to add the feature.

denravonska, 2024-01-11:
I have something that works but isn't ideal. The encoded size for C is done in `ProtoFile.generate_header` where it injects its dependencies into `Message.encoded_size`. The C++ part happens in `Message.fields_declaration_cpp_lookup` which doesn't have access to the ProtoFile dependencies.

As a test I introduced a `Message.max_size` field which is populated during header generation and used in the C++ lookup generation. Far from neat :(

PetteriAimonen, 2024-01-11:
Could it generate a function or const variable that references the C `#define`?

denravonska, 2024-01-11:
> Could it generate a function or const variable that references the C `#define`?

That was one of my approaches, but the C define isn't always there and the C++ generator doesn't know :(

PetteriAimonen, 2024-01-11:
Hmm, `#ifdef` would be one approach. But I'm also open to some refactoring in the generator if needed.

denravonska, 2024-01-11:
I'll post a PR so you can have a look.

PR #933 (2024-01-11): Add message sizes to C++ wrappers.
This adds a max_size field referring to the C #define, indicating the maximum encoded size of that message type. If the size cannot be determined, for example due to non-bounded repeated fields, the constant is omitted.

If anyone knows how to test for absence of types in C++ I'm all ears :)

This closes #932
commit bfe9b688 (2024-01-11): Add message sizes to C++ wrappers.

This adds a max_size field referring to the C #define, indicating the maximum
encoded size of that message type. If the size cannot be determined, for
example due to non-bounded repeated fields, the constant is omitted.

This closes #932
closed 2024-01-11 by PetteriAimonen