nanopb/nanopb #934: NanoPb v0.4.8 is broken, options do not work

Xeenych, 2024-01-14:
I'm using nanoPb 0.4.8 to compile the message
```
message host_to_hub_t {
    option (nanopb_msgopt).submsg_callback = true;
    option (nanopb_msgopt).type = FT_CALLBACK;
    oneof message {
        hub_control_t hub_control = 1[(nanopb).type = FT_CALLBACK]; 
        mac_txdata_t mac_txdata = 2 [(nanopb).type = FT_CALLBACK]; 
    }
}
```

v0.4.8 does not generate callback types for fields.

v0.4.7 does

PetteriAimonen, 2024-01-14:
The example you have provided does not compile for me on either version, because the referenced types are not included.

PetteriAimonen, 2024-01-14:
Possibly related to this other weird regression problem: #923.

Xeenych, 2024-01-14:
Yes it is.
Options do not work

PetteriAimonen, 2024-01-14:
You need to provide a buildable example if you want me to try it.

Xeenych, 2024-01-14:
I've updated the code to include full source of .proto file

PetteriAimonen, 2024-01-14:
After adding the missing `import "nanopb.proto";` that gave an error message, it produces identical output on 0.4.7 and 0.4.8 for me.

There is not much I can do about this until there is some way to reproduce the problem. You could try to debug it yourself, of course.

closed 2024-01-14 by PetteriAimonen

Xeenych, 2024-01-14:
Give it another try please. 
Now with russian comments. 0.4.7 produces output in windows-1251 encoding, but 0.4.8 produces utf-8 encoding. May be that somehow relates to missing field callbacks

```
syntax = "proto3";
import 'nanopb.proto'

// Сообщения от хоста к хабу
message host_to_hub_t {
 option (nanopb_msgopt).submsg_callback = true;
 option (nanopb_msgopt).type = FT_CALLBACK;
    oneof message {
        hub_control_t hub_control = 1;  // управление хабом
        mac_txdata_t mac_txdata = 2;    // данные с MAC уровня. Напрямую уходит в передатчик
    }
}

// Сообщения от хаба к хосту
message hub_to_host_t {
    oneof message {
        status_t flow_control = 1;      // управление потоком
        mac_rxdata_t mac_rxdata = 2;    // данные с MAC уровня, полученные приемником
    }   
}

// Управление потоком
// Чтобы избежать переполнения буфера хаба, после каждой команды `host_to_hub_t` хост должен дождаться ответа `status_t` с результатом выполнения команды
enum status_t {
    ok = 0;     // ok - команда выполнена успешно
    error = 1;  // какая-то ошибка
}

// Принятый пакет с MAC уровня
message mac_rxdata_t {
    int32 rssi = 1;             // rssi в dBm
    int32 snr = 2;              // snr в dBm
    uint32 sensor_address = 3;  // Адрес датчика, с которого пришел пакет
    bytes raw_data = 4;         // Сырые данные application уровня
}

// Пакет для передачи на MAC уровень
message mac_txdata_t {
    uint32 sensor_address = 1;  // Адрес датчика, на который надо отправить пакет
    bytes raw_data = 2;         // Сырые данные application уровня
}

// Настройки хаба
// Требуется дальнейшее определение
message hub_control_t {
    
}
  
```

PetteriAimonen, 2024-01-14:
The changed encoding is an expected difference, documented in [changelog](https://github.com/nanopb/nanopb/blob/master/CHANGELOG.txt#L4)